opentx/opentx #2372: no SWR on my Taranis B01

diablol, 2015-06-19:
opentx 2.1

After some openrcforums post I start a request here.

I have a Taranis B01 and have SWR with any 2.0.x version. But NO SWR valus with 2.1 that not seems normal!? 

Is it known? I found no information.

bsongis, 2015-06-19:
You did well in opening a new issue, I can't read forums as much as I should.

diablol, 2015-06-25:
Would be interessting someone else to confirm this. There is a minor possibility my board could be broken. I am getting a new one no later than next week.

projectkk2glider, 2015-06-25:
Confirmed on Taranis B01 with DR8 receiver (hub protocol). I will take a look at this...

closed 2015-06-25 by projectkk2glider

diablol, 2015-06-27:
so, means closed it should work now? because it doesnt.

projectkk2glider, 2015-06-27:
Yes, it should work. Did you look on telemetry setup page, the SWR will be the last item.

tomkfly, 2015-06-27:
The early B01 Taranis Plus TXs didn't have the hardware for SWR.

diablol, 2015-06-27:
I have no Plus and HAVE SWR with 2.0. 
checked again but nothing..:(

projectkk2glider, 2015-06-27:
Do you have connection with the receiver?

diablol, 2015-06-27:
Ah ok, I need connection. I see. 
But why? Should SWR not be independet from a reciever signal?

projectkk2glider, 2015-06-27:
It is sent from XJT internal transmitter to the radio CPU along with other telemetry data. That only happens when connection with the receiver is established.

diablol, 2015-06-27:
But that changed from opentx 2.0 where SWR worked without connection.
I am not sure if this is the best solution because you get a SWR warning first if your already connected.
If it cannot be changed with the new 2.1. concept it is still well paid.

projectkk2glider, 2015-06-27:
The swr warning is independent of telemetry settings, it will still  sound as before in 2.0

commit 260da330 (2015-06-25): Fixes #2372: SWR telemetry value not available on Taranis