opentx/opentx #2373: 9xr pro firmware upgrade / downgrade

drones-squad, 2015-06-19:
Hello,

I wanted to try the openTX firmware on my NEW 9xr pro radio.
However, since I have wanted to update the firmware with the original ersky firmware didnt work to well.

I have choosen the correct settints and downloaded the latest companion software. 
Before i writed the openTX tothe radio I made sure I would save the original backup rom from the radio transmitter.

When I wanted to instal this back , lost all my settings and kept saying strange things at first.
(error firmware) Then after shutting it down for the 2nd time and switching it on , did work.
This unstable kind of programming made me very nervous and suspicious to trust this at all ...

ALSO were ARE the voice commands ?? Why isn't this as easy as the ersky ?
I mean ok i get it. Its a diy upgrade , and respect for the people who made this. 
But isnt it time that someone should start working more on the GUI ? or make it more userfriendly ?
All that voice command stuff and mixing codes and LUA coding etc etc. 
Why  isnt this in one single firmware ?  I just dont see the point why EVERY people should read tons of information first in order to make it work as long as a TEAM already did that in the first place when they made it available out there for us. Thx but not convenient enough to use it as a starter.

greets

bsongis, 2015-06-19:
1) ersky9x settings are not compatible with opentx settings, so a format is normal.
2) the voice commands, did you install some sounds on your SD? If no you can download them from Companion
3) Work on the GUI, don't hesitate to give an hand if you have ideas and time to help, this is designed how we like it!
4) LUA on sky9x is not available (too little RAM available) You need a Taranis for LUA scripts.
5) Read docs, yes, there are tons of information on the internet, videos etc.

closed 2015-06-19 by bsongis

kilrah, 2015-06-20:
ersky9x and OpenTX are 2 firmwares made by different people with slightly different visions on how to best do things.
Just choose the one that works best for you.

drones-squad, 2015-06-20:
Hello,
However there are still issues with the companion regarding the 9xr pro. Do you guys have something planned for the near future ?Also does this support only the sam3 cortex cpu or others as well ?I was thining in replacing the mpu with a drop in replacement mpu working the same as the oem one but allot faster and powerfull .Thsi would benefit the special programmable functions and scripts. Thats why open tx is a must but not convenient at the moment.It would have helped many peoples and starts in the near future if it had a good gui and user friendly even dummies manuals.
greets