opentx/opentx #2374: latest Simu openTx:  Trim ON shows allways extened trimm valus, Extendet Trim is OFF

hrenz, 2015-06-20:
latest openTx from 19.06.  Simu
At Simu,  Show trim ON
Now allways trim is from +/- 125% possible
Extended Trim is NOT aktivated!

[image]
[image]

kilrah, 2015-06-20:
Following https://github.com/opentx/opentx/commit/f4e0518cd929fc0e078f8ea53693fd86e3ae1c06 trims are simply no more displayed in % but in steps.

closed 2015-06-20 by kilrah

bsongis, 2015-06-20:
Right, with percent, it was not possible to read the current trim value, as the precision was lost. It is steps, it could perhaps be percent with precision 1 instead.

kilrah, 2015-06-20:
> it could perhaps be percent with precision 1 instead

I think I tried that when I previously fixed an issue with that trim display, but it would have required rework of the display routines as the TINSIZE handling of decimals took too much space.