opentx/opentx #2378: Companion import of a 216 eeprom is broken

kilrah, 2015-06-22:
Invalid data + crashes

bsongis, 2015-06-22:
Would you attach an eeprom?

kilrah, 2015-06-22:
http://www.andrebernet.ch/divers/eeprom-good-2015-06-19-16-42-47.zip

I was looking at P-51, flight modes tab crashes

projectkk2glider, 2015-06-28:
The switch import is still wrong, the added SF- is causing problems. It probably should be fixed in `class SwitchesConversionTable` but it is hard for me to decipher what to do there.

fraternl, 2015-06-28:
My EEPE's can't be imported either. I'm assuming it's the same bug...
Should I post something or just wait and test after some fix is made?

bsongis, 2015-06-28:
Yes please attach your eeprom here and tell us what goes wrong, thanks!

projectkk2glider, 2015-06-28:
@fraternl be sure to use the latest nightly build before posting new problems.

bsongis, 2015-06-28:
I will wait for some tests before closing this one.

projectkk2glider, 2015-06-28:
Still some problems with switches

In Companion 2.0.17:
[image]

In latest `next`:
[image]

bsongis, 2015-06-28:
I knew it wouldn't be so simple :)

projectkk2glider, 2015-06-28:
It is good now!

kilrah, 2015-06-28:
Discussed via chat, but for the record still module settings broken and input names missing.

bsongis, 2015-06-28:
Input names are now good, is there anything else missing?

kilrah, 2015-06-28:
Looks all good now!

bsongis, 2015-06-28:
Cool!

commit e4b14233 (2015-06-28): Should fix #2378 - Who can confirm?
closed 2015-06-28 by bsongis