opentx/opentx #2379: 2.017 and Norton

HairyHugh, 2015-06-23:
Norton anti-virus reports a problem with this release and instantly deletes it. 2.015 was OK

projectkk2glider, 2015-06-23:
Must be something on your side. The report of the official download http://downloads-20.open-tx.org/companion/companionInstall_2.0.17.exe looks OK: 

https://www.virustotal.com/en/file/fe96664fa36b02347ce49ccb9d30b790d1384e2b52db0f2d98b6a8bc6a0a4bef/analysis/1435066322/

closed 2015-06-23 by projectkk2glider

HairyHugh, 2015-06-23:
Not my side, it is Symantec and Norton anti-virus that does not like it 
for some reason. I believe you can submit it to them for further 
analysis which should cure the issue.

Hugh

Madness helps

"You are not entitled to an opinion unless you can state the arguments against your opinion better than your opponents can."
Charlie Munger

For everyone's PC and internet security please delete all names and email addresses if you are forwarding messages.

kilrah, 2015-06-24:
Norton is a plague that should be eradicated ;)

HairyHugh, 2015-06-24:
Safest anti-virus on the block. Possible a bit over careful, but better 
that way than infected. :-)

As a retired guy who was involved in secure work I prefer to be a bit 
paranoid.

Madness helps

"You are not entitled to an opinion unless you can state the arguments against your opinion better than your opponents can."
Charlie Munger

For everyone's PC and internet security please delete all names and email addresses if you are forwarding messages.

griebd, 2015-06-24:
@HairyHugh, sorry to say, and completely of topic, but: DON'T use windows if you are just a bit careful! someone that uses windows saying he is a bit paranoid is almost hilarious! (and don't come with the stuff that you can't do something, there are virtual machines for ages now....)

kilrah, 2015-06-24:
> (and don't come with the stuff that you can't do something, there are virtual machines for ages now....)

But then you're just running Windows again ;)

Seriously... I've been running multiple Windows machines without antivirus for 10 years now. And Norton is pretty much the worst bloatware of all time and usually causes more problems than it solves ;)

HairyHugh, 2015-06-24:
Not really helpful. As someone who started with a Commodore Pet back in 
the 1970s I have used multiple systems from there through VAX VMS, UNIX 
and Microbloat. I have yet to find a decent system, they are all c**p in 
my humble opinion.

As for antivirus I have seen too many people have multiple problems 
because they were not using any. Perhaps they were just unlucky and 
naive, but they still got screwed.

If you choose not to address the issue with Norton (which I agree is 
bloated, but it is effective and nothing like as bad as all the UNIX 
implementations) then I choose not to use 2.017 as 2.015 does all I 
need. Perhaps we should then leave it at that and part on amicable terms.

Regards

Hugh

Madness helps

"You are not entitled to an opinion unless you can state the arguments against your opinion better than your opponents can."
Charlie Munger

For everyone's PC and internet security please delete all names and email addresses if you are forwarding messages.

projectkk2glider, 2015-06-24:
Then at least provide us with the output of your antivirus and a hash value of your downloaded file that causes this.

Have you checked my second link? According to Virustotal the Companion is in the clear.

jtaylor2, 2015-06-24:
Actually the link shows that everybody shows it clean except Symantec and they show  as detecting WS.Reputation.1 which means they will quarantine the file.  WS.Reputation.1 isn't really a virus, it is Symantec being evil and reporting something because it is new or not widely used.  See http://community.norton.com/en/forums/clarification-wsreputation1-detection for their explanation of WS.Reputation.1.

Symantec is will known for being bloated and for false positives.  Symantec detected every release of the SeaMonkey browser as having some kind of virus and the team submitted samples before the release trying to get them to correct the problem, but they would never do it in any kind of a timely manner.  Finally they gave up and put it in the release notes http://www.seamonkey-project.org/releases/seamonkey2.33/ : 

>Norton/Symantec anti-virus scanners may report that some parts of SeaMonkey (e.g. the file >freebl3.dll) are suspicious. If you downloaded SeaMonkey from one of the official download sites, >this is a false alarm. You might experience problems with secure websites when this happens. To fix >the issue, instruct your anti-virus softwaret to ignore these files (and move them out of quarantine) >and/or switch to another anti-virus software and reinstall SeaMonkey.

@HairyHugh as a Symantec user you should report the false positive to Symantec along with your displeasure with them quarantining files just because they are not widely used, if in fact you are displeased.

Now for my off topic soapbox I worked in IT for 45 years,  I have never ran an antivirus on any of my 8-10 systems and I have never had a virus, maybe I've been lucky, but I help my luck by being careful.  However, I do recommend people that aren't careful run one, but I recommend that they don't run Symantec because it is a resource hog and generally causes people more problems than it prevents.  And as a last word, if you have something really important and you are really worried about security it need to be on a mainframe.

Jim

kilrah, 2015-06-24:
There's a saying in IT circles that the more you pay for an antivirus the more false warnings you're gonna get. Cause obviously to the average user if nothing was ever reported they'd realise it's useless and stop renewing their subscription... gotta wake up that sense of insecurity regularly enough to keep the business going.

Open source software is often targeted for reasons as above of "nor widely used" or "not using common (paying) distribution channels". 
Some time ago the dfu-util binary that was bundled with companion triggered a false warning from some AVs on the basis of "low level access". Well no shit, that's what it's made for...