opentx/opentx #2382: Vario only showing positive values

helicop, 2015-06-24:
As title. Taranis Plus. Nightly 23/06/2015. Frsky D4R-11 and OpenXVario. In sensor page (13/13) only positive values are shown for Vspd.

projectkk2glider, 2015-06-25:
My solution in https://github.com/opentx/opentx/commit/a88cace88855aed107dd79f4b4559de9220cac4c was wrong on two accounts: a bug and wrong solution. It has been reverted.

@bsongis @kilrah Now let's discuss how to solve this problem:

The setting to zero was done because in gtest we do not allow negative value for current sensor. 

Obviously we need an option that prevents some sensor values from becoming negative. Ones such case is the current sensor. Other sensor values might also need this (maybe).

There are several options to solve this:
* we go with the current solution and we add if() case for VSPEED (like my attempt, but with corrected condition) 
* we remove this feature altogether - all values can be negative
* we introduce a new setting for each telemetry value that defines if value can be negative.

kilrah, 2015-06-25:
I don't see why we disallow negative values, so I'd remove the check altogether...

commit a88cace8 (2015-06-25): Fixes #2382: Vario only showing positive values
commit 56c24932 (2015-06-25): Fixes #2382: Vario only showing positive values (reverted from commit a88cace88855aed107dd79f4b4559de9220cac4c)
commit 7312bd26 (2015-06-28): Fixes #2382: new atribute onlyPositive for telemetry sensor

Conflicts:
	radio/src/translations/it.h.txt
closed 2015-06-28 by projectkk2glider