opentx/opentx #2385: [Enhancement] Avoid Companion to overwrite "persistent timer"

LapinFou, 2015-06-25:
Hi Folks,

I've recently modify a model from Companion. When I transfered the new model in my Taranis, I realized that all my persistent timers have been overwritten by the value stored in my one month old EEPE.
Not a big deal, but I open this ticket to propose an enhancement: would be it possible to add an option in Companion to NOT overwrite "persistent timers"?
In other world, when transferring  a modified model, the timer value which must be kept is the one stored in the Taranis, not the one stored in the EEPE.

kilrah, 2015-06-25:
If you do it "correctly" i.e. load eeprom from radio, modify, then rewrite timers should be unchanged.

closed 2015-06-25 by kilrah

LapinFou, 2015-06-25:
I don't have the feeling to have done it "incorrectly":
1. I know no settings have been changed in the radio.
2. I opened a backup EEPE from my USB key at work.
3. I've done some modifications during my lunch time at work.
4. Back at home, I transfered the new EEPE from my USB key to my Taranis.
5. Everything was fine (thank the Companion simulator).
6. I lost my persistent timers.

It was just a suggestion.
;-)

kilrah, 2015-06-25:
You would also lose any trim changes done in the meantime.

I really recommend never to reuse "old" files for anything other than recovery in case you inadvertently lost or made destructive changes to a model. Always read straight from the radio before making any change, or even better back up the radio after each flight session. That way you're sure you're always working on the latest data set and you always have it at hand.

LapinFou, 2015-06-26:
Arrggg !! I also forgot about the trims (which certainly slightly changed) !!
:]
You're absolutely right. Thanks for taking the time to explain me how wrong was my procedure.
:+1: