opentx/opentx #2393: Support for OrangeRX 2.4Ghz module in 9XR-PRO

DavidHache, 2015-06-29:
OpenTX doesn't seem to support the recent module "OrangeRX 2.4Ghz Telemetry Module For Turnigy 9XR Pro (DSMX/DSM2/Walkera Compatible)".

With ersky9xr R217, it is supported when putting the protocol to 9xr-dsm.

Is there a plan to integrate or a way I may help integrate this module?

bsongis, 2015-06-29:
What do I need to buy to make it work?

ckleanth, 2015-06-29:
Bernard have a read on these posts
http://www.rcgroups.com/forums/showpost.php?p=31898169&postcount=2351
http://www.rcgroups.com/forums/showpost.php?p=31902129&postcount=2354
cheers

MikeBland, 2015-07-03:
This is the module: http://www.hobbyking.com/hobbyking/store/__61213__OrangeRX_2_4Ghz_Telemetry_Module_For_Turnigy_9XR_Pro_DSMX_DSM2_Walkera_Compatible_.html.
It uses the internal antenna of the 9XR-PRO. If the PCB is placed in a different case, with an external antenna in a suitable place, this does plug into a Taranis.
The telemetry data is sent on pin 5 of the module connector, NON-inverted. The 'PRO handles this easily as we have a programmable hardware inverter on that signal.
For the Taranis, a software driven serial input is needed. I have one working, and have just got support for this module working in ersky9x on the Taranis. The software serial is handled by an interrupt when the signal changes state. For this to work reliably, it needs to be the highest priority interrupt as it needs to read a timer value to capture the time of the state change. We don't have a timer capture directly available on this signal. In ersky9x, I now set the priority of every interrupt, and use 0 (highest) for this, and something lower for all others.
I also have a DSM telemetry module from another manufacturer, not yet commercially available. The code to handle this module is similar, although this module uses inverted serial data, and the SPort signal is used for both control and telemetry. While adding support for the Orange module, it would be useful to add support for this module at the same time.
I will be committing my code changes for ersky9x soon, but I'm busy with several things and also need to migrate to Github.

Mike.

bsongis, 2015-07-15:
Which is the receiver you recommend I buy? I will only use it for dev/test, I don't use DSM2 on my models.

MikeBland, 2015-07-15:
Tricky. The module handles both DSM2 and DSMX and 11mS and 22mS frame rates for various different receivers. I've tested Lemon DSM2 both 6ch and 10ch, Orange R615X (DSMX) Spektrum AR6210, and the Lemon stabiliser receiver. None of these do telemetry directly.

Perhaps the Orange R720X or R820X would be suitable as these have integrated telemetry, although I haven't got either. Someone had a problem on the forum, but does have a 720X working with telemetry. The problem was when binding, you can disable telemetry by having an extra bind plug on the throttle channel. It seems a servo plugged in was detected as a bind plug!

My code is now on Github: https://github.com/MikeBland/mbtx.

Mike.

bsongis, 2015-07-18:
No stock for the module :(

DavidHache, 2015-07-18:
I have the module, is there a way I can help you test it?

D

bsongis, 2015-07-18:
Ok I will add the feature and you will test, not easy, but no choice. It would be nice if all vendors could send their hardware for which we add support... this is what FrSky does and it's really easier.

DavidHache, 2015-07-18:
Let me know what version has the module so I can test it.

andyjenkinson, 2015-07-24:
@MikeBland what is the pinout like? Thinking it is a shame if the internal Taranis antenna cannot be used.

MikeBland, 2015-07-24:
The internal antenna on the Taranis is hardwired to
the internal XJT module and cannot be used by any external module.

Andy Jenkinson wrote:

  @MikeBland what is the pinout like? Thinking
it is a shame if the internal Taranis antenna cannot be used.
  —
it on GitHub.

kilrah, 2015-07-24:
... And no radios other than the 9xr/9xr pro are able to make use of the module and it's proprietary antenna connection without serious mods.

andyjenkinson, 2015-07-25:
Yeah I noticed the odd antenna connection, I just thought the taranis antenna was broken out to the module bay connector. Shame that it's hard wired to the XJT, the extra antenna on these modules has always set off my OCD ;)

kilrah, 2015-07-25:
The problem is that in 2.4GHz you can't just use a simple pin like back in the days in the MHz range. That's why the 9XR has that coax connector added, but they're the only ones to have that.

disodbediente, 2015-07-27:
9XR pro wasn't a big deal after all. I have to modules and none yet works (more specifically I'm unabile to use them, no info, no manual, nothing). Btw I come here to ask: This orange module says it's compatible with walkera. Are they saying that trasmits also in DSSS? I'm unable to bind :-(

With DSM2/DSMX it works, thanks to all the info I found online. Doesn't work with original firmare. But binding walkera devo…??

I also have a DSSS module from turnigy for 9XR pro… it was supposed to be perfect for walkeras protocol. Never had a sign of life from it.

Argh!
If I put together how many hours I spent… probably it was enought to buy an Inspire 1.

xLORDxSIDIUSx, 2015-09-04:
i really want to purchase this module.  but i use open TX... not willing to switch to ersky9x.  has progress been made for Open TX support?  if i buy this module, is there a version i can test?

bsongis, 2015-09-05:
Yes support will be added in OpenTX, I am still waiting HK have them in stock ...

DavidHache, 2016-06-16:
Hi, 
Any news from HK for the module?
I would love to fly with OpenTX instead of the stock firmware.  Much easier to configure.

kilrah, 2016-06-16:
I believe HK discontinued that module right after the first batch, never saw it available again and apart from you I don't know any customer who has it...

bsongis, 2016-06-16:
I can't have one myself :( 
I was said my other Orange Tx module could be re-flashed, I will try when I have one minute

timemaster5, 2016-07-23:
So is there any update ? I have one module too. So can I help somehow ?

schwabe, 2016-07-25:
The best way to help is to write the code. The "normal" dsm code might be adaptable for the 9XR-DSM protocol. Even documenting the protocl would be a step forward. (the ersky9x code is a bit messy there). 

I have already implemented a telemetry parser for the Spektrum telemetry that may need some minor modifications for that module. (At the moment it expects the header that the MULTI code uses).

The problem that no one of the regular OpenTX programmers has access to the module and so the priority of implementing something you cannot test yourself is probably very low for most (speaking for myself here mainly). 

Another option would to flash the MULTI code on that module (http://openrcforums.com/forum/viewtopic.php?t=8753) but the multi implementation is missing the menu item ons the 9XR and the serial code isn't tested.

MikeBland, 2016-07-25:
I would suggest heading down the route of flashing the MULTI code to that module. The firmware in the module as shipped has the odd flaw, as I discovered when porting the MULTI code.

timemaster5, 2017-02-23:
Mike, thanks for this solution. I have successfully flashed my OrangeRX module with https://github.com/pascallanger/DIY-Multiprotocol-TX-Module by your XMEGA flasher inside ersky9x FW on my Turnigy 9XR PRO and it worked great on first try. With this simple solution, I am able to use my module either with OpenTX (patched for multi-protocol support) or ersky9x, without strugle... I think that this is much better solution than trying to fix communication issues especially if original firmware has some bugs inside..

dieideeistgut, 2017-05-10:
@timemaster5 any chance you could sum up the steps you've done to accomplish this?

closed 2017-11-10 by kilrah