opentx/opentx #2399: [Enhancement] Filtering 6 position switch

LapinFou, 2015-07-01:
Hi folks,

I installed a 6 position switch in my T+ in place of my S2 pot. Since this 6 position switch uses a resistor scheme, the correct detection of S21, S22, ... S26 is ensure.
However, if you are moving from one position to another you can observe a glitch. This is due to how FrSky designed this 6 position switch.
I was planning to use this 6 position switch to commute my flight modes. But, because this glitch, I will not (too risky).
Could it be possible to implement a glitch killer? Something similar to the "Play Delay (sw. mid pos)" feature (Page 1/8 in Radio Setup)?
It could be something hidden in the FW or maybe the user could setup the "switch delay" as it is done for "play delay".

kilrah, 2015-07-01:
Already done if you set the pot type to multipos switch on the hardware tab and calibrate.

closed 2015-07-01 by kilrah

LapinFou, 2015-07-01:
I've already done the setup "multipos switch", otherwise I would not have been able to select S21, S22, etc...
I think the filtering windows is definitely not long enough, because I can easily get a glitch specially when switching from S21 to S22.
I'm running OpenTX 2.0.17
I've got 2 T+ with the same behavior.
Could you please have try?

bsongis, 2015-07-01:
Would you try to change your "Play delay" in Radio Setup?

LapinFou, 2015-07-01:
Didn't know that this parameter was also impacting the 6 position switch.
I change the value from 150ms to 300ms (which is the maximum). The glitch issue is still happening (less often, but still present).
I think for the 6 pos sw, you need a delay between 500ms et 1s.
In the setup I tested, S21 will stop the engine (security position). But, in order to initiate the engine, I must put back the throttle to 0 (in fact -100%) to start the engine.
When I switch from S22 to S23, the switch glitch-ed to S21 and stopped my engine. I take me sometime before understanding what was going on. Fortunately I was flying a glider, not an helicopter.
;)

LapinFou, 2015-07-05:
Thanks !!! :+1: 
Any chance to include this fix in the version 2.0.18?

bsongis, 2015-07-05:
Well, when the rush is gone, yes.

bsongis, 2015-07-05:
Did you test it on 2.1.0? Because if it doesn't fix your problem, it won't need to be ported...

LapinFou, 2015-07-05:
I just tried on 2.1.0. Still the same issue, BUT I think the "Play Delay" is not applied to S2 (Multipos switch).
I can observe the glitch directly on the main menu, just here:
[image]

LapinFou, 2015-07-05:
Because with 1s, the glitch should definitely not happen.

LapinFou, 2015-07-05:
By the way, there is something which I don't understand about S25 & S26 value:
S21 = -100
S22 = -60
S23 = -20
S24 = 19  ??? why not 20 ???
S25 = 59  ??? why not 60 ???
S26 = 100

LapinFou, 2015-07-05:
I tried again => stupid me !!!
OK the slider on the snapshot below is not filtered, BUT the usage of S2x is correctly filtered.
So, it is working, but maybe the display should also be filtered (only when "Multipos switch" is selected) to avoid any confusion.

commit 34240377 (2015-07-01): Fixes #2399