opentx/opentx #2400: Failsafe Mode not shown correctly in Companion to 2.1.0

RC-SOAR, 2015-07-02:
Currently testing Companion 2.1.0

I am loading a setup (see link below) which has Failsafe Mode set to 'custom'

1. Opened setup in Companion 2.1.0 
2. In the Setup tab, Failsafe mode is shown as 'Not Set'
3. In the simulator, at startup displays a 'failsafe not set warning

.EEPE file is here:
https://drive.google.com/file/d/0B7eJi6dZb_UkaHRHQ1M0bnZGbE0/view?usp=sharing

I haven't tested it on a transmitter yet.

kilrah, 2015-07-02:
2.1.0 of which day? Things were fixed very recently about failsafe.
Where does this eepe come from? What version of companion was it made with?

RC-SOAR, 2015-07-02:
I downloaded and installed OTXC 2.1.0 approx 20 minutes ago using downloads-21 link.
The .EEPE was originally created in r2940, eventually translated to format 216 EEPROM after release of OpenTx 2.0. Full change log here:
http://rc-soar.com/opentx/setups/f3f/changelog.htm

RC-SOAR, 2015-07-02:
Additional info: I always set the failsafe via the Tx, not via Companion. Using X8R rx.

closed 2015-07-02 by kilrah

MIKE-DAILEY, 2015-07-02:
Because of the new features added in the "custom" failsafe, it seems like the failsafe will need to be set again after the 2.1.0 upgrade for each model for users using eepes made on 2.0.17 and below that had "custom" failsafe set?  

Mike

commit 07ac7db0 (2015-07-02): Failsafe mode was converted wrong when coming from a 216 eeprom (Fix #2400)