opentx/opentx #2401: Heli menü in Companion V2.10 and openTX is different

hrenz, 2015-07-02:
At companionV2.10 Heli menü is not complete, 3 Weight missing
different to openTx V2.10

kilrah, 2015-07-02:
???

[image]

closed 2015-07-02 by kilrah

bsongis, 2015-07-02:
Perhaps he means the order of fields? Which one is the best, this is easy to fix ...

kilrah, 2015-07-02:
We can at least change the labels so they match the radio... "elevator" and "aileron" sucks for helis :)

bsongis, 2015-07-02:
So we keep the radio order, ok?

hrenz, 2015-07-02:
think Radio order is ok and fine to set for Heli

bsongis, 2015-07-02:
You may download the latest companion installer, it is fixed.

hrenz, 2015-07-02:
Thanks a lot!
I am just making translations ready for companion and opentx and testing

commit 1ad54b24 (2015-07-02): Fixes #2401