opentx/opentx #2402: Error writing setting to radio  -  Companion 2.1 /  9XR PRO

procha170, 2015-07-02:
Hello,
I tested firmware and companion 2.1   on my radio 9xr pro. When I want to transfer setting into radio I get error popup window

Error reading file c:\ ... AppData/Local/Temp/9340-temp.bin

Not allowed access to memory place (translation from Czech)

Reading data from radio works normally
 
I'm able open temp file from error message in companion but I'm not able to write it into radio.

I have Windows 7 64 bit / SP1

digifaps, 2015-07-22:
I Have the same issue.

same on win 8.1 and win 7 here.

Tried it on several machines to no avail.

please fix as we can't use open-tx like this.

same behaviour :

can flash firmware
can read settings 
can restore backups
CANNOT save the read settings back to the same radio.

bsongis, 2015-07-22:
Duplicate of #2517

closed 2015-07-22 by bsongis