opentx/opentx #2403: Differential behaviour changes following migration to 2.1.0

RC-SOAR, 2015-07-02:
Diff is not working the same way after migrating my F3F setup from 2.0.15 to 2.1.0.  This is both in Companion and on the actual model.

Diff on the right aileron appears to work correctly. However on the left aileron the diff is now affecting the up-going aileron instead of the down-going aileron. 

Reproducable on both the Tx and the Companion 2.1.0 (1st July build). 

Steps to reproduce:

1. Download https://drive.google.com/file/d/0B7eJi6dZb_UkaHRHQ1M0bnZGbE0/view?usp=sharing
2. Load in OTX Companion 2.1.0
3. Open model 01
4. Open sim view in editor
5. move SA to the middle (Normal mode)
6. Hold full left aileron (RH stick), vary diff with rudder trim. Note that neither aileron moves.
7. Hold full right aileron, vary diff with rudder trim. Note that BOTH ailerons move.

Loading the same model def in OpenTx Companion 2.0.15 works as expected.

RC-SOAR, 2015-07-02:
OK here's a minimal setup, should be easier to work with

https://drive.google.com/open?id=0B7eJi6dZb_UkX0pBZ05SX2Zsek0

Aileron diff value is stored in GV1, which is bound to CH14 (rudder trim)

kilrah, 2015-07-02:
Hmm not sure why that changed, but from what I remember it's the old way that would likely be "wrong"... I remember when we introduced GVs there that we allowed choosing -GVx as it would be required precisely in that scenario.

bsongis, 2015-07-02:
Right I thought we didn't change anything, something must be wrong, I will find tomorrow.

kilrah, 2015-07-02:
This?

[image]

bsongis, 2015-07-02:
Here:
https://github.com/opentx/opentx/commit/f008e6ad889834b74085dbec6caec09f1ac9258a
So this was intentional, to fix this issue:
https://github.com/opentx/opentx/commit/f008e6ad889834b74085dbec6caec09f1ac9258a

bsongis, 2015-07-02:
@JivaroFAD I need advice from more gliders experts ;)

RC-SOAR, 2015-07-02:
Advice is simple: backwards compatibility.

bsongis, 2015-07-02:
Perhaps I will revert as we have bigger plans on differential with @JivaroFAD

RC-SOAR, 2015-07-02:
Just to clarify: I am not affected by #1879 as (a) I do not have mixer offsets and diff in the same mix and (b) I always split trim into a separate mix. So maybe I'm the wrong person to ask. All that concerns me is that users should not have to re-adjust their diff settings, or alter program settings for these cases.

bsongis, 2015-07-03:
I don't agree here, it would mean that if we did something wrong in a release N, we cannot fix in a release N+1?

On the differential topic, our implementation is not the best, I really want to sort it as soon as possible. But introducing a partial change in 2.1.0 doesn't seem good, I want to have things done properly if we do them.

kilrah, 2015-07-03:
No, what he says is we shouldn't change something if not necessary and if it might change again later, i.e. we all agree.

So yes, revert the commit, "unfix" #1879, and just make sure to take it into account when reworking differential "properly".
BUT, on that topic I believe that differential rework should target a major version, i.e. either delay 2.1 to include it or postpone it to 2.2. We can't change basic model behavior in a minor release, people wouldn't expect something as fundamental to change there.

bsongis, 2015-07-03:
Ok. Then it will be done in 2.2.0, I mean next major release. We have changed enough things in this release, at least the mixer will stay untouched.
Give me 1/2h it will be ok.

closed 2015-07-03 by bsongis

bsongis, 2015-07-03:
Please download again the 2.1 companion and firmware, this one is reverted:
http://downloads-21.open-tx.org/companion/

RC-SOAR, 2015-07-03:
Have just installed the latest Companion (3rd July), have tried clicking on 'Check for Updates' but no firmware is shown. Any clues?

bsongis, 2015-07-03:
Yes, we don't increment the number in this phase, while the release is not official just install again as if you never did it.

RC-SOAR, 2015-07-03:
Confirmed - diff behaviour in latest build has reverted to as it was in 2.0.15. Many thanks!

commit 22f598cd (2015-07-03): [Taranis] Fixes #2403 - Differential changes we did in 2.1 reverted