opentx/opentx #2404: OpenTx 2.1 enhancement for model-name.wav

rdeanchurch, 2015-07-02:
I always have the model name spoken as well as displayed on the tx.  It would be very convenient if a model_name(.wav) could be included on the setup page just as the model image(.bmp) is.

Then a Global function to play model-name or just automatically played or with a paly/no-play checkbox.

deaconFlyer, 2015-07-03:
I would love to see a file name "model-name.wav" be part of a "aircraft-name" directory in the sounds directory. I have found this to be the quickest way to maintain my setup and switch sounds. If the name was present, the firmware would speak it when selected.

You would not need a global function after that. If it is present, it is spoken,

You could have some combination of both items.

kilrah, 2015-07-03:
> I would love to see a file name "model-name.wav" be part of a "aircraft-name" directory in the sounds directory.

Doesn't that exist already?

projectkk2glider, 2015-07-05:
No, after looking at the sources, there is no such function implemented (yet).

rdeanchurch, 2015-07-07:
As Global Functions appear to be processed prior to the data in Setup of a model, I prefer having a global function have a method define a .wav for a model name.  
As opposed to having the model name being buried ion any global function Play Tracks that might be present.  
 I assume a Sound/Model folder solution is also processed after global functions, but have not tested.

kilrah, 2015-07-07:
Global functions can not have any model-specific action. The sound in the model folder is pretty much the only valid solution.

PR #2501 (2015-07-19): Fixes #2404: model name played when model is loaded
Plays `SOUNDS/…<language>/<model_name>/name.wav`

Closes #2404
commit 08d4cb15 (2015-07-19): Fixes #2404: model name played when model is loaded (if file "SOUNDS/<language>/<model_name>/name.wav" exists)
commit 1492d808 (2015-07-19): Merge pull request #2501 from opentx/projectkk2glider/issue_2404_play_model_name

Fixes #2404: model name played when model is loaded
commit c0fbd5cc (2015-07-19): Fixes #2404: model name played when model is loaded (if file "SOUNDS/<language>/<model_name>/name.wav" exists) (ported from master and squashed)
closed 2015-07-19 by projectkk2glider