opentx/opentx #2409: [OpenTX2.1.x][Enhancement] Adding global logical switch

LapinFou, 2015-07-03:
Having some global functions is (once again ;-) ) a very very very nice feature.
It would be perfect to also have some global logical switches (ex: to setup a advanced throttle security switch combined with some global functions).

projectkk2glider, 2015-07-05:
Needs eeprom change, so it will have to wait until 2.2

LapinFou, 2015-07-05:
OK.
Thanks!
:]

GilDev, 2018-02-26:
Up for that. I have a special combination of logical switches and functions that allows me to read or reset my timers when holding a lever for different amounts of time. It would be great if I could configure it once for all my models instead of redoing everything each time.

RipperDrone, 2019-12-23:
I have a low battery warning feature implemented based on latched logical switches (voltage <x for y seconds --> battery warning), even refined it to two alarm levels (bat low, bat critical). 

Now I have 50+ models defined on the Taranis X9D+, but the battery voltage alarm is identical throughout all of them. I would LOVE to be able to handle it with global functions / global logical switches so each model can reuse the same battery warning feature without having to code it on 50+ models separately, which is what I have to do now. 

Any chance of adding the global logical switches feature into 2.4 maybe? :-)

3djc, 2019-12-23:
logical switches are not the issue, sensors are defined per model

GilDev, 2020-06-20:
Oh no! 🥺

LapinFou, 2020-12-18:
I hope not!
😕

0815Creeper, 2021-06-08:
Um, there are not many features that are still missing in otx but this is definitly one of them...

0815Creeper, 2022-04-16:
No, still to relevant to be stale!

Kjblootens, 2023-08-06:
Can we define global logical switches, based only on radio-based inputs (e.g. sticks, switches and pots) that can be used within models as logical values.

So we would have GL01-GLXX which I can then use in my models to affect a standard radio configuration.

This would avoid re-creating such logical switching in each model. The GLXX vs LXX naming avoid confusion within models programming.

An example would be a set of 6 GL linked to the 6 positions of the 6Px using a~x and [-100,-60,-20,20,60,100], or a set combination of switches as a pre-arm condition.