opentx/opentx #2410: Special Function "reset Alt" does nothing

fraternl, 2015-07-03:
I just connected my DLG with vario.
I'm running version 2.1.0 (downloaded using Companion 2015-07-03 on that same date.

It gives me a starting altitude of -61 metres...
If I do a telemetry reset by press and holding the button right under on the Taranis and select telemetry reset it stays that same value

If use the special function "reset Alt" it stays the same.
I can't get it to 0 metres....

I am just getting acquainted with the new system and am trying to find out everything intuitively.
Maybe I'm not doing it as I should

kilrah, 2015-07-03:
Check "Auto Offset" in the sensor's settings.

closed 2015-07-03 by kilrah

fraternl, 2015-07-03:
Yes, that was it....
Thanks