opentx/opentx #2411: Reading altitude gives me the height, but it says "miles per hour" instead of metres

fraternl, 2015-07-03:
If I do an altitude reading on my Taranis which is set in "metric", it will give me "miles per hour" as unit instead of metres.

I know I'm not using the Amber sound pack anymore because it's a robot-like voice.

2.1.0 2015-07-03

kilrah, 2015-07-03:
You need to use the new 2.1 sound packs. http://voices-21.open-tx.org

fraternl, 2015-07-03:
I thought I already had the 2.1 sounds.
To make sure I just downloaded the en file from that site and copied all sounds again to my Taranis.
It did the same...

I then downloaded the scottish-english pack and the french pack.

If I set my radio to French it says "metres"
If I set it to English it continues to say "miles per hour"

Now to really make sure I only have the 2.1 sounds (no hybrid with the Amber sound-pack) I renamed my "en" folder to "en-amber" and extracted the "english-american-taranis.zip" directly to the SD-card thus creating a new "en" folder.

It still says "miles per hour"
Could you please check it?

fraternl, 2015-07-03:
I can't explain it what I just wrote...
Just checked file /SOUNDS/fr/SYSTEM/0133.wav and it says metres in French and so does it in English....

But it still comes out as "miles per hour" when I set it to English (as if plays 0132.wav)

Are these the files we are talking about?

kilrah, 2015-07-03:
Looks like there's a shift of 1 in units in English. Don't know if it's in the file numbering or in the speech script yet.

fraternl, 2015-07-04:
That is indeed strange.
BTW....  to find the file I had to try the wav files one by one because the file system.txt in /sounds/ wasn't helping me.

fraternl, 2015-07-04:
I just replaced the file 0132.wav (which is supposed to say "miles per hour") with a copy of 0134.wav
When I now do a "Play value Alt"  it says "metres".
I don't know if it is supposed to play file 0133.wav or 0134.wav, but I prefer the plural

This is all very very strange as it correctly says "volts" when I do a "Play value RxBatt"

closed 2015-07-04 by bsongis

bsongis, 2015-07-04:
I made a new release, you should install it over the previous one, use also the new firmware and download the new sounds!

Please tell-me if things are right, thanks!

fraternl, 2015-07-05:
Hi...
I just replaced the file 0132.wav to "miles per hour" and checked "play value Alt"
It plays "miles per hour" when I do that....
Note that it plays "metres" if I put it in French....

To make sure I just played these sound files directly from the USB-attached Taranis.
I found some more anomalies in the French directory (might be more than this)

--------------- English ------------------- French
0123.wav --- kilometer per hour ---  midi???
0124.wav --- kilometers per hour --- missing
0131.wav --- mile per hour ----------  kilometre heure
0132.wav --- miles per hour --------  miles per heure
0133.wav --- metre ------------------- metre
0134.wav --- metres ------------------ metre(s)

kilrah, 2015-07-05:
File numbers are different for each language, there is specific code for each.

fraternl, 2015-07-05:
I just now noticed there were some changes in the sound files.
I downloaded these....

French is still botched....

--------------- English ------------------- French
0123.wav --- metre per second --- midi???
0124.wav --- metres per second -- missing
0131.wav --- metre ---------- kilometre heure
0132.wav --- metres -------- miles per heure
0133.wav --- foot ------------------- metre
0134.wav --- feet  ------------------ pi (???)

fraternl, 2015-07-05:
"File numbers are different for each language, there is specific code for each."

Ah, ok.....
Assumed wrong...
But why do it that way? Things like this are then more likely to happen.....

kilrah, 2015-07-05:
Again there is separate code for each language with different file ordering, it's normal that different files are played for the same thing.

Please try the latest firmware and latest packs for each language without messing with files and report any errors.

kilrah, 2015-07-05:
> But why do it that way?

Because some languages use different logic, have more/less words... for example some languages like Czech have 4 different words for each unit instead of just 2 for singular/plural.

fraternl, 2015-07-05:
"Please try the latest firmware and latest packs for each language without messing with files and report any errors."
I did this in the process...
I will put the new French on the radio, but I need to rework the English because I want the Amber sounds and have to check it all again which takes some time

fraternl, 2015-07-05:
"Because some languages use different logic, have more/less words..."
OK

After reworking the system sound pack to "Amber" I will check some more....
I do hope the file dates will help me to see what has been changed....

Will report later.
I guess you tested it already and it's probably good now

kilrah, 2015-07-05:
Do not think about Amber right now. Please see if the packs we distribute work to confirm all is good with the fixes that have been made. THEN once that's good you can use it as a reference of how you must rename the Amber pack files. 
Otherwise you might be doing the work for nothing, as if things are not good names might change again.

fraternl, 2015-07-05:
All seems to work properly....
Keep the ticket closed...
I did do the Amber conversion, but will triple check with the original set if I think something is off......

bsongis, 2015-07-05:
Would you provide a link to your Amber pack modified? I think it's the users' preferred one. Thanks

fraternl, 2015-07-05:
http://hmvc.eu/Amber.2.1.System.Sounds.zip

But there's still something wrong with minutes and seconds

"2 minutes and 4 seconds" is spoken as "2 seconds and 4 ..."

kilrah, 2015-07-05:
Redownload firmware, it was fixed in the meantime.

fraternl, 2015-07-05:
Yes... Downloaded firmware again and it's now fine  ;-)

Here's a link to the Amber thread which has the 2.0 complete set.
http://www.rcgroups.com/forums/showthread.php?t=2151914
My patch needs to go over that one

bsongis, 2015-07-05:
So now it works with the sounds inside your link, right?
All units tested? Play Duration tested?

fraternl, 2015-07-07:
During the week I have no access to my radio, so I can't test any further. I think everything is working properly and no files needed to be moved to get it working.

Will we be using Amber for English in the future...

I'm considering a sound pack for Dutch. I just now spoke to someone with a studio. I also need to find someone to speak the voices. It will be a lot of work.

Is there even an option for Dutch?

bsongis, 2015-07-07:
You could create a tts_du.cpp from tts_en.cpp, plus the prompts, and I will integrate it.

fraternl, 2015-07-07:
I had a look and I could start on an "nl.h.txt" which I would base on "de.h.txt"
The voice pack would take some time.
Is it possible to already have the code provisioned?
I might be able to have that "nl.h.txt" in a reasonably short time.

I only own a Taranis plus.
Could this be a problem for the other flavours?

bsongis, 2015-07-07:
Yes it's a good idea to start with the translations for screen!

fraternl, 2015-07-08:
I am editing de.h.txt to create nl.h.txt and I am using en.h.txt as a refefence....

At the position of the 2nd "#if defined(VIRTUALINPUTS)" in en.h.txt, the German equivalent has "#if defined(PCB_TARANIS)" with other text

kilrah, 2015-07-08:
English is usually the reference, when things are changed they are done there and it is up to the translators to follow and replicate them which they don't always do.

fraternl, 2015-07-08:
"You could create a tts_du.cpp from tts_en.cpp, plus the prompts, and I will integrate it."

Is there some way to make tts.nl.cpp a softlink to tts.en.cpp as they are the same.
That way a change in tts.en.cpp will reflect in the Dutch translation.

I already have a Dutch soundpack, but it's incomplete. If I merge it with the 2.1 Amber set will that be a problem?
I mean... would you accept it as a Dutch pack?

bsongis, 2015-07-08:
I prefer we keep both separated. Ok for the sound pack, you know better than me anyhow!

fraternl, 2015-07-08:
http://hmvc.eu/SOUNDS_nl.zip
Here's my Dutch soundpack

http://hmvc.eu/tts.nl.cpp

commit c05dd459 (2015-07-04): Fixes #2411