opentx/opentx #2412: Editing a curve point can cause instability

RC-SOAR, 2015-07-04:
OpenTx v. 2.0.1 (3rd July build)
Taranis Rev B01

Odd stability issue when editing a curve point. It's happened with a couple of different models. Symptoms vary from non-working sound to permanent corruption of the model definition.

Steps to reproduce:

1. From the bootloader, restore and convert this 216 format EEPROM
https://drive.google.com/file/d/0B7eJi6dZb_UkSEpSZ01UUU9uM00/view?usp=sharing
2. Select model 09 'CULAST_03'
3. Open Curves menu and edit CV4 'Thr2Sp'
4. Highlight Pt 2->X and [Enter]
5. Keep finger pressed on + key 

At this point, there's a click, and OpenTx jumps to the main info screen.
Other symptoms vary. Sometimes the sound ceases to work for the rest of the session. At others, the CULAST_03 model definition is permanently corrupted in the EEPROM (saved example:
https://drive.google.com/file/d/0B7eJi6dZb_UkVVBKQmhMOWRqS1k/view?usp=sharing )

A similar issue can occur when editing the corresponding curve point in STING_J2.

bsongis, 2015-07-04:
This one is a no go for X9E / Taranis 2.1.0!

bsongis, 2015-07-04:
Reproduced on simu, it should be easier to fix than on the radio itself. I will be on it in 1 hour.

bsongis, 2015-07-04:
Inside drawFilledRect!

commit 6ca949d0 (2015-07-04): Fixes #2412
closed 2015-07-04 by bsongis