opentx/opentx #2413: 2.1.0 - model image selector possibly broken

lichtl, 2015-07-04:
I've updated for testing/review of the new telemetry to 2.1.0 on my Taranis. During model setup I've noticed that the model image selector is not generated properly (I'm uploading a video on YT right now). It looks like the right border is being created but it don't overwrite the other sections.

lichtl, 2015-07-04:
This is the video. I've just reviewed the play-track selector. This is working as expected.

https://www.youtube.com/watch?v=HLMMLJ34cyg

kilrah, 2015-07-04:
How many files do you have in there?

I see an issue as well but a different one...

lichtl, 2015-07-04:
I have 523 files in the BMP folder. This is a copy. On 2.0.17 and below I never experienced any issues here.
https://drive.google.com/file/d/0B3ugUwyDKJageVdtX3NnNnIzT0E/view?usp=sharing

bsongis, 2015-07-04:
Exactly the same on the simu and your directory

commit 4c4d3bc2 (2015-07-04): Fixes #2413
closed 2015-07-04 by bsongis