opentx/opentx #2414: Companion - switches not in parameter list

Mow06, 2015-07-04:
reported here for 2.0.17.  2.1 seems the same.

http://www.rcgroups.com/forums/showpost.php?p=32065754&postcount=9053

I am using a Switch as its own parameter in Special Functions.

That is, a Special Function is that when SF is Up it sets the Volume to the value of SF (which is -100%)
A second Special Function SF Up is also my Throttle Kill switch for electrics.

The result is it stops any announcements if the model is disarmed. Shuts the Taranis up while you are setting it up or adjusting stuff. 

However I notice I cannot set this up in Companion as the switches themselves do not appear in the Special Function drop down list of Parameters. However, I can do it on the transmitter where switches do appear in the parameter field and it then works fine. If I then read it back into Companion the Parameter simply appears as ....

Anyone know why this is so? Is it just a missing function in Companion or is it done some other way in Companion.

I'm using Companion 2.0.17 Mac version.

Thanks for any insight.

John

HotTaurian, 2015-07-04:
If you are only talking about setting up a throttle kill switch
add a "thr" with a curve and all points set to -100 activate with the switch of your choice on your radio.
Set up your curve first, then add the additional "thr" line
this addition is done in the mixes tab.
I tried to add this type of option in the inputs tab and it kills all inputs on a T-9XR Pro.
I hope this helps those who haven't figured it out.

jj604, 2015-07-04:
Thanks, but that's not the issue. I'm the person who commented on it in the RCGroups thread.

The issue is that Companion has a different set of parameters than on the Taranis in Special Functions. The physical switches are not available in the drop down. However they can be set on the transmitter and are preserved by Companion when it reads the transmitter.

commit 0783f728 (2015-07-04): Fixes #2414
commit 396ded7e (2015-07-04): Fixes #2414
closed 2015-07-04 by bsongis