opentx/opentx #2415: Random default source when creating new mix

RC-SOAR, 2015-07-04:
OpenTx 2.1.0

1. Create a new model
2. Open the mixer editor in an empty channel

The default for the mixer source seems to be random, in one mix case it defaulted to '[C1]' or '[C3]' neither of which appears in the source pick list.

kilrah, 2015-07-04:
It's pretty much a feature, a new mix on CH1 takes the first source list entry, a new mix on CH8 takes the 8th, etc.

RC-SOAR, 2015-07-04:
Yep, the pseudo-random bit is fine (I guess you have to start from somewhere)... what concerned me was that in a couple of instances it chose defaults of "[C1"] and "[C3]" which didn't appear in the picklist (accessed using Enter and +/- keys). 
Background to this is that I'm attempting re-building one of my setups from scratch through the transmitter UI.

RC-SOAR, 2015-07-04:
Ah, a clue... these show as 'CYCn' in Companion, so I guess these refer to heli inputs. Just looking now, the CYC entries show up in the source picklist in the Companion editor, but not in the Tx.

kilrah, 2015-07-04:
CYC1-3 are the CCPM mixer outputs. They will show as [Cx] on the radio if you have the noheli option selected as that mixer is then not present.

If you have a noheli FW I think everything is OK.

bsongis, 2015-07-04:
Sources which are not available should be skipped, that's right. I will fix it.

commit 017c13db (2015-07-04): Fixes #2415
closed 2015-07-04 by bsongis