opentx/opentx #2416: All subtrim values are off after migration

fraternl, 2015-07-04:
I flew a few days ago with a MiMi, which is a 2-axis mini DLG and noticed that trimming was somewhat off, but because the thing is not that precise I didn't think much of it...

Today I flew 2 of my F3K DLG's and they have very precise controls on the aileron.
Both my DLG's were off with their subtrim.

The values themselves migrated properly, but the ailerons were clearly in the wrong position.
I needed to recalibrate.

I'm using the 2015-07-03 version and the conversion was done on 2015-07-03

kilrah, 2015-07-04:
If you use differential there's a change that was reverted in #2403 and might affect you. Was done yesterday, so your test is likely obsolete already... see how it works now.

fraternl, 2015-07-04:
I do use differential and will load tomorrow's firmware to my radio and will report if this behaviour reverted

fraternl, 2015-07-05:
This morning I downloaded the new Companion and firmware and checked one of my DLG's
Its ailerons are at the same position I trimmed them yesterday.
I would have expected the subtrim to be off again....

BTW If I check the timestamp of the firmware using radiosettings it says: 2015-07-05  09:53:54
I downloaded it approximately at 07:50 GMT+2
I assume the version I pick is compiled on-the-fly on the server if it does not exist.already.
It takes an hour to be 09:53 in my timezone and even 2 hours more if it's supposed to be GMT.

Or is it some Eastern European local time?

bsongis, 2015-07-05:
Did you provide your EEPROM? I will need it to compare. Please tell me which model and which channel needs checking.

fraternl, 2015-07-05:
Hi Bernard,
I saved all the models in an .eepe before I migrated. I then saved them again after migrating (before flying) and I can save them now (after I recalibrated 3 models in there).
I will start compiling them now

bsongis, 2015-07-05:
I will need the eepe before the conversion + information to find as quickly as possible where your problem is.

fraternl, 2015-07-05:
[image]
It's a zip-file

kilrah, 2015-07-05:
> information to find as quickly as possible where your problem is.

Please ;)

RC-SOAR, 2015-07-05:
Just wondering, did you check PPM offsets were the same?

kilrah, 2015-07-05:
And did you do the conversion with companion or on the radio?

RC-SOAR, 2015-07-05:
Or could it be a stick calibration issue (e.g. if using 2 tx's or maybe factory reset)?

kilrah, 2015-07-05:
I tried opening the 2.0.17 eeprom in companion 2.0 and the one labelled as "directly after migrating" in companion 2.1 and on model MiMi I can see a difference in subtrim on CH1 (-2.5% instead of-5.2%).

If I however open the 2.0.17 eeprom in today's companion 2.1 or run the conversion using today's firmware simulator both conversions are OK and the simu outputs are identical. So either it's not "directly after migrating" or the conversion was done with a version that had a bug that was fixed in the meantime.

When you report a bug please always repeat the procedure on the very latest version even if you just did it a day ago and make sure the problem is still there.

Closing this for now, reopen if you can still reproduce the problem and point to an exact place where it's apparent.

closed 2015-07-05 by kilrah

fraternl, 2015-07-05:
I think I will revert to 2.0.17 load that 2.0.17 eepe and check my settings. Then do an upgrade to the current firmware.
I let my radio do the upgrade and flew that Mimi afterwards. The "Cumel" and "JS" were the models where I noticed it.
In calibration mode there's no diff so I don't understand how that could be of influence.

Anyhow....
I will also write down the output values in both 2.0.17 and 2.1.0

And I will check calibration values....

fraternl, 2015-07-05:
I reverted to 2.0.17 loaded the eepe and checked a specific model. (JS).
I aligned the tips of the ailerons with the wing and then I did a firmware upgrade.
Everything stayed the same...

So, keep it closed. 
Sorry for any inconvenience