opentx/opentx #2418: UI error in Special Functions

RC-SOAR, 2015-07-04:
Taranis, OpenTx 2.1.0 (3 July build)

The new 'source category' dropdown in 2.1.0 is great, but it doesn't appear to work correctly when creating an 'Adjust GV' special function. 

Steps to reproduce:

1. Go to Special Functions
2. Create a new function with action 'Adjust GV1'
3. Skip to source field, and [Enter long]  repeatedly until the source category picklist is displayed.
4. Select a group e.g. 'Sticks', and press [Enter]

At this point, one would expect to see the field value default to the first value in the group, but it always shows 'GV1',

kilrah, 2015-07-04:
Ah yes there is a conflict here between the ENTER LONG that is used to toggle the source type only for this particular SF, and the ENTER LONG that calls the source popup in all source fields.

Hmm, what's the best way to solve that? Can't think of anything good.

bsongis, 2015-07-04:
Good catch. I propose to remove the source category in this choice (Adjust GV), ok?

RC-SOAR, 2015-07-04:
So it'll work the same as before? Sounds fine.

bsongis, 2015-07-05:
I keep it open, but delayed to 2.1.1 for the nice fix ...

kilrah, 2015-07-05:
Oh, easy... We also have an issue open for +/-x instead of only 1... can probably see that at the same time.

commit 6374f525 (2015-09-06): Fixes #2418
commit 64bdface (2015-09-06): Merge pull request #2822 from opentx/bsongis/issue2418_better_fix

Fixes #2418
closed 2015-09-06 by bsongis