opentx/opentx #2419: Negative range limits for bar graph telemetry display

Novgorod, 2015-07-05:
The limits for the bar graph telemetry display cannot be negative - this produces bad behavior with telemetry values which can become negative (such ass AccX, Y, Z, which are displayed fine in a text-only screen). The filled portion of the bar graph should start at 0 and fill up to the telemetry value to the right (positive) or to the left (negative), as it is implemented in the channel monitor. This sanity check (i.e. the lower limit must be below the upper limit) is already implemented, as well as out-of-range-behavior, so I don't see a reason to limit the displayed range to nonnegative values.

Oh and while you're at it, is there a reason to limit the range for AccX, Y, Z to 2.55? If I'm not mistaken, the protocol supports greater g-values than 2.55, which are also displayed as text just fine - only the bar graph range is limited without ovious reason...

bsongis, 2015-07-06:
If I am not mistaken this issue is in 2.0.17, but not 2.1.0: in 2.1.0 there is no such limit (2.55) and the minimum can be negative

Novgorod, 2015-07-07:
Ok, I din't check 2.1.0, only the latest public release, which is 2.0.17 (and didn't find anything related in the bug tracker)...

closed 2017-11-29 by mpaperno