opentx/opentx #2420: [2.0.17] OpenTx companion crash on linux when editing input

eephyne, 2015-07-05:
When I go to a model to edit the input (maybe mixers too but since input doesn't work…) companion crash

I'm on linux (arch, kernel 4.0.5 x86_64)
companion is set for 9x (I have a flysky th9x).

the stack trace is :

*** stack smashing detected ***: companion terminated                                                                                                            [116/121]
======= Backtrace: =========                                                                                                                                     [115/121]
/usr/lib/libc.so.6(+0x71bad)[0x7f8211e74bad]                                                                                                                     [114/121]
/usr/lib/libc.so.6(__fortify_fail+0x37)[0x7f8211efaff7]                                                                                                          [113/121]
/usr/lib/libc.so.6(__fortify_fail+0x0)[0x7f8211efafc0]                                                                                                           [112/121]
companion(_ZN11InputsPanel11gm_openExpoEi+0x235)[0x5f72c5]                                                                                                       [111/121]
companion(_ZN11InputsPanel8expoOpenEP15QListWidgetItem+0x7b)[0x5f79bb]                                                                                           [110/121]
companion[0x61f202]                                                                                                                                              [109/121]
/usr/lib/libQtCore.so.4(_ZN11QMetaObject8activateEP7QObjectPKS_iPPv+0x4d0)[0x7f8213314e20]                                                                       [108/121]
/usr/lib/libQtGui.so.4(_ZN17QAbstractItemView13doubleClickedERK11QModelIndex+0x35)[0x7f82142f8015]                                                               [107/121]
/usr/lib/libQtGui.so.4(_ZN17QAbstractItemView21mouseDoubleClickEventEP11QMouseEvent+0xb5)[0x7f8214301905]                                                        [106/121]
/usr/lib/libQtGui.so.4(_ZN7QWidget5eventEP6QEvent+0xe20)[0x7f8213e16540]                                                                                         [105/121]
/usr/lib/libQtGui.so.4(_ZN6QFrame5eventEP6QEvent+0x1e)[0x7f82141dbb9e]                                                                                           [104/121]
/usr/lib/libQtGui.so.4(_ZN17QAbstractItemView13viewportEventEP6QEvent+0x1e3)[0x7f8214301443]                                                                     [103/121]
/usr/lib/libQtCore.so.4(_ZN23QCoreApplicationPrivate29sendThroughObjectEventFiltersEP7QObjectP6QEvent+0x86)[0x7f8213300eb6]                                      [102/121]
/usr/lib/libQtGui.so.4(_ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent+0x6c)[0x7f8213dbf11c]                                                           [101/121]
/usr/lib/libQtGui.so.4(_ZN12QApplication6notifyEP7QObjectP6QEvent+0x476)[0x7f8213dc6156]                                                                         [100/121]
/usr/lib/libQtCore.so.4(_ZN16QCoreApplication14notifyInternalEP7QObjectP6QEvent+0x8d)[0x7f8213300d4d]                                                             [99/121]
/usr/lib/libQtGui.so.4(_ZN19QApplicationPrivate14sendMouseEventEP7QWidgetP11QMouseEventS1_S1_PS1_R8QPointerIS0_Eb+0x14d)[0x7f8213dc575d]                          [98/121]
/usr/lib/libQtGui.so.4(+0x245742)[0x7f8213e40742]                                                                                                                 [97/121]
/usr/lib/libQtGui.so.4(_ZN12QApplication15x11ProcessEventEP7_XEvent+0x5fd)[0x7f8213e3f10d]                                                                        [96/121]
/usr/lib/libQtGui.so.4(+0x26d082)[0x7f8213e68082]                                                                                                                 [95/121]
/usr/lib/libglib-2.0.so.0(g_main_context_dispatch+0x24d)[0x7f8211b3e9fd]                                                                                          [94/121]
/usr/lib/libglib-2.0.so.0(+0x49ce0)[0x7f8211b3ece0]                                                                                                               [93/121]
/usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x2c)[0x7f8211b3ed8c]                                                                                          [92/121]
/usr/lib/libQtCore.so.4(_ZN20QEventDispatcherGlib13processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE+0x64)[0x7f8213331044]                                 [91/121]
/usr/lib/libQtGui.so.4(+0x26d156)[0x7f8213e68156]                                                                                                                 [90/121]
/usr/lib/libQtCore.so.4(_ZN10QEventLoop13processEventsE6QFlagsINS_17ProcessEventsFlagEE+0x41)[0x7f82132ff5c1]                                                     [89/121]
/usr/lib/libQtCore.so.4(_ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE+0x1b5)[0x7f82132ff935]                                                              [88/121]
/usr/lib/libQtCore.so.4(_ZN16QCoreApplication4execEv+0x99)[0x7f82133052e9]                                                                                        [87/121]
companion(main+0x6cc)[0x4ce58c]                                                                                                                                   [86/121]
/usr/lib/libc.so.6(__libc_start_main+0xf0)[0x7f8211e23790]                                                                                                        [85/121]
companion(_start+0x29)[0x4cfeb9]                                                                                                                                  [84/121]
======= Memory map: ========                                                                                                                                      [83/121]
00400000-00c00000 r-xp 00000000 08:02 504141                             /usr/bin/companion                                                                       [82/121]
00c00000-00c01000 r--p 00800000 08:02 504141                             /usr/bin/companion                                                                       [81/121]
00c01000-00c05000 rw-p 00801000 08:02 504141                             /usr/bin/companion                                                                       [80/121]
00c05000-00cb4000 rw-p 00000000 00:00 0                                                                                                                           [79/121]
02711000-03b0d000 rw-p 00000000 00:00 0                                  [heap]                                                                                   [78/121]
7f81d8000000-7f81d8021000 rw-p 00000000 00:00 0                                                                                                                   [77/121]
7f81d8021000-7f81dc000000 ---p 00000000 00:00 0                                                                                                                   [76/121]
7f81dc000000-7f81dc021000 rw-p 00000000 00:00 0                                                                                                                   [75/121]
7f81dc021000-7f81e0000000 ---p 00000000 00:00 0                                                                                                                   [74/121]
7f81e0000000-7f81e0021000 rw-p 00000000 00:00 0                                                                                                                   [73/121]
7f81e0021000-7f81e4000000 ---p 00000000 00:00 0                                                                                                                   [72/121]
7f81e6021000-7f81e6036000 rw-p 00000000 00:00 0                                                                                                                   [71/121]
7f81e6122000-7f81e616d000 rw-p 00000000 00:00 0                                                                                                                   [70/121]
7f81e6223000-7f81e6245000 rw-p 00000000 00:00 0                                                                                                                   [69/121]
[... cut ...]
7f81ee9e8000-7f81ee9e9000 rw-p 00007000 08:02 543696                     /usr/lib/qt4/plugins/imageformats/libqjpeg.so
7f81ee9e9000-7f81ee9f0000 r-xp 00000000 08:02 543702                     /usr/lib/qt4/plugins/imageformats/libqico.so
7f81ee9f0000-7f81eebef000 ---p 00007000 08:02 543702                     /usr/lib/qt4/plugins/imageformats/libqico.so
7f81eebef000-7f81eebf0000 r--p 00006000 08:02 543702                     /usr/lib/qt4/plugins/imageformats/libqico.so[1]    7845 abort (core dumped)  companion

I currently try to build it from git to see if there is a difference

bsongis, 2015-07-05:
Would you check with Companion 2.1.0?

eephyne, 2015-07-05:
I just did (wanted to build master version but forgot to switch branch).

build version is 2.0.99 and it work perfectly (for now) :)

So…Can I use it without too much worries of bug or is it really unstable ? (still haven't updated radio for now)

bsongis, 2015-07-05:
2.0.99 is out of date for sure!

eephyne, 2015-07-05:
why its this version who was built using the next branch ?

kilrah, 2015-07-05:
You likely did not "git pull --rebase" to get the changes that have been made since you last checked out next.

eephyne, 2015-07-05:
exactly @kilrah, I forgot that the repo was sync a long time ago.

I'm in 2.1.0 now and it work still well

I ask again, does 2.1.0 (and more generally next branch) is usually good to go or too buggy overall ?

kilrah, 2015-07-05:
2.1 is currently in the final beta stage with release hopefully in a few days. See the open and closed issues on the 2.1 milestone.

eephyne, 2015-07-05:
Ok

Since it final beta I assume you will not search for this bug in 2.0.17.

If this is the case, I'll mark it as solved, otherwise I'll keep it open

projectkk2glider, 2015-07-05:
No, keep it open. I will have a look..

projectkk2glider, 2015-07-05:
I found it :beetle: . It is also present in `next`. 

It's a sneaky bug that was easy to fix (fix coming soon). Here is an example scenario that produces this bug:

We are editing 9X radio so input names are not used, the relevant code:
```
void InputsPanel::gm_openExpo(int index)
{
    // code snipped
    char inputName[4+1];
    // code snipped
    ExpoDialog *g = new ExpoDialog(this, model, &mixd, generalSettings, firmware, inputName);
    // code snipped

```

So, the `ExpoDialog` constructor gets a pointer to un-initialized `inputName` buffer in this case. In a release build this buffer has a random value, the trouble happens if it contains a value that resembles an UTF8 encoded string without null termination.

The  `ExpoDialog` class uses this pointer and sets a text edit control value to this string. The control has a text length limit set to 4 characters, but if this characters happen to be UFT8 encoded then we can have a buffer overrun:

For example for `inputName  = "ćžšn"` the `strlen(inputName.toAscii().data())` is `7 `! 

This `strcpy` does the buffer overrun:
```
void ExpoDialog::valuesChanged()
{
    // code snipped
    strcpy(inputName, ui->inputName->text().toAscii().data());
    // code snipped
```

commit 8d96cfa4 (2015-07-05): Fixes #2420: input name buffer overflow in Companion
commit c09d4278 (2015-07-05): Fixes #2420: input name buffer overflow in Companion (ported from master)
closed 2015-07-05 by projectkk2glider