opentx/opentx #2421: Radio didn't turn off after using the switch

fraternl, 2015-07-05:
After flashing the 2015-07-05 firmware and checking some settings I wasn't able to turn off the radio. I waited more than 2 minutes. In fact I'm still waiting whilst writing this message.
It's on "Saving model settings"

I can't say if this is a fluke or if it happens on more occasions.
I never had it before. I do hope it doesn't corrupt my models if I unplug the battery....

bsongis, 2015-07-05:
Instant-on is a feature, yes! In case your radio reboots when flying, you don't want to be blocked on an EEPROM / THR / SW warning, the SD is not mounted to avoid any problem in this area as well.

bsongis, 2015-07-05:
Now your problem. In OpenTX 2.1 we use hardware I2C instead of software I2C, it now goes around 5 times faster to read / write the EEPROM. Would you do more tests, you just modify something, shutdown.

kilrah, 2015-07-05:
I've had mine lock at the same moment once a couple of days ago. Happened only once, but with your report there might be something.

bsongis, 2015-07-05:
Taranis Plus for both reports?

kilrah, 2015-07-05:
For me yes

RC-SOAR, 2015-07-05:
Same issues as the o/p here (need to disconnect battery to shut down), but with an earlier build.

Possibly related to this, I've had a couple of freezes when scrolling up/down the model list. Can't remember the exact details, but I do remember the channels continued to respond correctly.

Tried to reproduce the issue with later builds, but unlike the o/o have not been able to do so. But it's been on my mind.

bsongis, 2015-07-05:
I succeeded to reproduce it. Also in the MODELSEL screen.

fraternl, 2015-07-05:
Yes, Taranis plus

bsongis, 2015-07-05:
I reproduced only one time ... I changed trims and copied the current model to another slot. The radio hanged (not exactly the radio, but the menus task, the mixer is still alive, I am almost sure we are blocked inside an I2C wait).

bsongis, 2015-07-05:
Now I reproduced 3 times ... I change trims and copy the current model to another slot in the same second or so. The radio hangs (not exactly the radio, but the menus task, the mixer is still alive).

The common function called in all our scenarios is eeCheck(true), pretty sure we are blocked somewhere inside, perhaps inside an I2C wait.

All ideas / other tests welcome, I am sure it's a firmware bug introduced in 2.1.0

bsongis, 2015-07-05:
Now inside DELETE MODEL

bsongis, 2015-07-05:
Still not good, I just had the problem on model delete

closed 2015-07-06 by bsongis

bsongis, 2015-07-06:
In order to test my fix, I put an alert(...) just before I2C reinitialization to be aware I would be in an endless loop before (I mean without my fix). And yes I had the alert, the I2C was reinitialized and the write continued where it was. So I am confident, this issue is gone :)

RC-SOAR, 2015-07-06:
That's great work - it sounded to me like it would be a tricky one to solve!

projectkk2glider, 2015-07-07:
@bsongis Is this a I2C speed issue or is it caused by the fact that we now use hardware driver I2C?

kilrah, 2015-07-07:
I would think it's related to the speed that might be borderline in some cases, that's the only thing I could think of that could cause problems...

bsongis, 2015-07-08:
I am sure of nothing here, here are some possible causes:
- the EEPROM doesn't answer or SDA / SCL remain high => when I have more time, I need to check this with the logic analyser
- there is a timing problem, a task woken up during the I2C translation? It shouldn't be a problem, but ...
- perhaps there is an error detected by the I2C hardware layer, it is reported but we don't check it correctly 
- the STlib implementation of I2C is not perfect (many reports on the web for this)

projectkk2glider, 2015-07-08:
I've created a new ticket to remind us of this remaining issue.

commit 87f07801 (2015-07-06): Fixes #2421 - Full check of I2C errors / infinite wait. In case of
eeprom read/write error, the I2C is reinitialized and the operation
started again.
commit 01b0c015 (2015-07-06): Fixes #2421 - Full check of I2C errors / infinite wait. In case of
eeprom read/write error, the I2C is reinitialized and the operation
started again.