opentx/opentx #2422: [2.1.0] avrdude try to write bin file in root folder

eephyne, 2015-07-05:
I just updated to 2.1.0 and wanted to flash my radio during check avrdude complain:

avrdude: writing output file "/backup-2015-07-05-164800.bin"
avrdude: can't open output file /backup-2015-07-05-164800.bin: Permission denied
avrdude: write to file '/backup-2015-07-05-164800.bin' failed

I did not found a parameter to change the path so I assume we can't.

Maybe its a flag to set during build (if its the case it definitely should not be / by default)

the full log:

/usr/bin/avrdude -c usbasp -p m64 -P usb -U flash:r:/tmp/13883-flash-check.bin:r
=================================================================================

avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9602
avrdude: reading flash memory:

Reading | ################################################## | 100% 32.86s

avrdude: writing output file "/tmp/13883-flash-check.bin"

avrdude done.  Thank you.

=================================================================================
/usr/bin/avrdude -c usbasp -p m64 -P usb -U eeprom:r:/backup-2015-07-05-164800.bin:r
=================================================================================

avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9602
avrdude: reading eeprom memory:

Reading | ################################################## | 100% 1.16s

avrdude: writing output file "/backup-2015-07-05-164800.bin"
avrdude: can't open output file /backup-2015-07-05-164800.bin: Permission denied
avrdude: write to file '/backup-2015-07-05-164800.bin' failed

avrdude done.  Thank you.

=================================================================================

eephyne, 2015-07-05:
nevermind, I found the option…

but still, it should be something else than / by default

closed 2015-07-05 by eephyne

7immy, 2016-01-28:
I'm having the same issue here. Where is this option you speak of?
Thanks

projectkk2glider, 2016-01-28:
Settings -> Settings... -> Application Settings tab -> Automatic Backup Folder