opentx/opentx #2425: Companion 2.1 doesn't import internal module settings from 217 eeproms

kilrah, 2015-07-06:
External and trainer are OK, but trainer shows PPM settings for SBUS input.

[image]

MIKE-DAILEY, 2015-07-06:
Thank you Kilrah and Bertrand.  This must be the fix for the backup problem for the files showing protocol off that I emailed to Bertrand.

kilrah, 2015-07-10:
Title issue wasn't solved...

bsongis, 2015-07-10:
In fact yes, you had a bad eeprom conversion which was fix since that time. If you compile with option TARANIS_INTERNAL_PPM you will be able to setup the right module type

fraternl, 2015-07-10:
I have firmware Jul 10th of firmware and companion and still can't import radio settings with the correct receiver.
What I can gather from your posts this has been fixed 4 days ago.
I'm only using nightly compiles, so I can't follow your advice to Kilrah.

Do I just have to wait another day?

bsongis, 2015-07-10:
When did you do the 216 => 217 conversion?

kilrah, 2015-07-10:
It can't be a conversion issue because I set up a new model on the radio today (internal on D8, FW compiled 2 hours ago with no TARANIS_INTERNAL_PPM option) and load it into companion built right now from next the internal module is shown as OFF in companion.

bsongis, 2015-07-10:
oh ok :)

MIKE-DAILEY, 2015-07-10:
Not sure if this will help any but after my internal module type was fixed with your firmware Bertrand I made a backup and as you know and it saved the protocol (module type).  I'm now on build #31 and if I do a backup it shows the correct protocol.  Is this what you are discussing?

fraternl, 2015-07-11:
The last time I touched my radio was on Sunday July 5th.

On Friday (yesterday) I downloaded a new companion and flashed the daily of that day to my radio.

I didn't see any notice that it converted my eeprom to a newer version.
Is this something that has to be done manually?

Or do I need to go back to 2.0.17, load my old setup and then do an upgrade to today's nightly?

commit 5b567fc0 (2015-07-06): Fixes #2425
commit d4544f11 (2015-07-11): Fixes #2425
closed 2015-07-11 by bsongis