opentx/opentx #2427: Opentx 2.1 splashscreens

mhotar, 2015-07-06:


projectkk2glider, 2015-07-07:
Nice.

bsongis, 2015-07-07:
Nice!

mhotar, 2015-07-07:
I finally removed radio frames from splash screens.

[image]

fraternl, 2015-07-07:
I agree they look nice.
I'm sure we will have these in our next Companion...  ;-)

I'm not a dev. though

tomkfly, 2015-07-07:
But we don't have couloured screens.

fraternl, 2015-07-07:
Do you still have a monochrome monitor attached to your PC?

tomkfly, 2015-07-07:
Of course not, but I don't see an point with a nice screen on the sim when it's not available on the TX.

kilrah, 2015-07-07:
It's got nothing to do with the sim, it's the splash screen when you launch companion.

bsongis, 2015-07-07:
Thanks, really nice work!

bsongis, 2015-07-07:
May I ask to work with @romoloman for the X9E simu graphics, I would like it is consistent with your new splashes!

closed 2015-07-07 by bsongis

mhotar, 2015-07-08:
I am happy to help. @romoloman do not hesitate to send more details and instructions what you would like me to do.