opentx/opentx #2428: Jul7 nightly builds altitude limits.

rdeanchurch, 2015-07-07:
Trying to set a Logical for a>x    Alt    390 feet.
(I fly near airports, legal limit is 400 feet.)

In OpnTxComp. limit was 39.00 feet.
On transmitter limit was 300.00 feet.

Reading Tx back to PC (win8.1) 300 feet was read back into OpnTxComp..

[As an aside, Is precision of 2 decimals points appropriate for necessary for Alt?  I'm using FrSky High precision...not good to 2 decimals places in my opinion.]

bsongis, 2015-07-07:
Is it 2.0 or 2.1?

fraternl, 2015-07-07:
I think it's 2.1
I noticed the 2 decimal points too, but didn't think of reporting it.

The resolution is 10 cm AFAIK for the high precision vario.
So I agree with you that double precision is not appropriate.

rdeanchurch, 2015-07-07:
Its 2.1.  I didn't think of the possibility of using 2.0.1x nightlies, my error.

bsongis, 2015-07-07:
I don't get you, what are 2.0.1x nightlies? Is the issue present or no, and in which release?

rdeanchurch, 2015-07-07:
I don't know either what 2.0.1x nightly builds are, that is why I said

"Its 2.1"  as in 2.1.0 nightly builds.
Sorry for the confusion.

Is there a better place to have discussions on points that may just be a mis-understanding?
An RcGroups or  openrcforums thread?

For instance, so far, I haven't been able to get the firmware upgrade to work on the Taranis with 2.1.0 nightlies.  Too soon or different method or...?

kilrah, 2015-07-07:
You're talking of 3 different things in one issue and it's getting confusing.

- The alt precision. The protocol is defined for sending info with 2 decimals, and thus that's how it's displayed by default. The 2nd might not be useful but it doesn't hurt either. Change precision to hide it if you want.

- The ft/m import/export difference in companion. So is there an issue with that or not in the end? If yes how exactly?

- Firmware upgrades. No changes have been made here and it works for others - so yes see forums for troubleshooting and/or clearing misunderstandings. Open a separate issue if a problem can be confirmed through discussion with others.

bsongis, 2015-07-07:
Ok I close it, feel free to open issue per problem if any!

closed 2015-07-08 by bsongis