opentx/opentx #2432: Unsupported characters in the "Inputs" fields in Companion 2.1

mhotar, 2015-07-07:
Hello,
I would like to ask for some help. In the Companion CZ translations I have special characters in the stick names. When the Input names are created (during new model creation) from the stick name, they contain unsupported characters.
Is It possible to filter them out somehow, so the stick "Směrovka" is translated to "Smer" not "Sm?r"

Thanks.

[image]

hrenz, 2015-07-07:
Same problem at German with characters ä ö ü

'Elevator Stick is Höhe  or "Höh"  shown als H?h   or  as  H h

projectkk2glider, 2015-07-07:
I am not sure how (even if possible) to solve this one.

The special language specific characters are encoded on radio into certain 8bit values (lets call this OpenTX encoding).  But different languages use the **same code for different characters**.

So Companion must first know (when importing EEPROM) in which radio language the  eeprom was created/edited. Only then can it know how to convert from OpenTX encoding into UTF8 encoding.

The reverse must happen when EEPROM is saved. Based on known radio language the UTF8 strings are converted into OpenTX encoding. 

So we must create two functions that convert between OpenTX encoding and UTF8 encoding in both ways. But still the problem of unknown eeprom language remains.

----
On a related note: why is number of special characters `LEN_SPECIAL_CHARS` for certain languages different from the actual number of character in the font files? One example is DE which has 7 characters, but LEN_SPECIAL_CHARS = 6.

mhotar, 2015-07-07:
I would propose to simply enable to translate these strings separately from the main stick names to use names without special characters.

On the firmware side I have implemented It the same way, I use string definition for INPUTNAMES specifically
#define LEN_INPUTNAMES         "\004"
#define TR_INPUTNAMES          "Smer""Vysk""Plyn""Krid" 

This is implemented for CZ only, because I am not good at that "ifdefs" at all. If someone with skills can remake it to work for all languages, like "if defined TR_INPUTNAMES" it will be even better.

kilrah, 2015-07-07:
Or use a function that stores a mapping of each special char to a legal ones for the firmware language, e.g. \xyz (ö) -> o. Call that when defining an input name from a stick name, so for example in German Höh becomes Hoh instead of H?h. 

So one function with a different array per language in #defines and that's it.

mhotar, 2015-07-08:
Works great. Thanks

projectkk2glider, 2015-07-09:
With this solution, if one imports EEPROM from the radio that contains special characters and then writes back this eeprom (without editing names) they loose special characters!

bsongis, 2015-07-09:
No, the function is not called in this case. It is only used on new model creation (from Companion)!

closed 2015-07-09 by bsongis

projectkk2glider, 2015-07-09:
@bsongis true, but the issue I described remains. I just tried it. Created a new EEPROM in DE simu, opened in EN Companion and saved back. When opened again in simu, the special characters in default input names are gone!

commit aac73844 (2015-07-07): Fixes #2432 - Perhaps some chars missing?