opentx/opentx #2434: Taranis Plus - Play Delay (sw.mid pos)

MIKE-DAILEY, 2015-07-07:
2.1.0 - Play Delay (sw.mid pos) does not seem to work anymore. I believe it was set at 150ms in 2.0.17 and I ran it all the way to 1000ms and it still plays the mid switch wav sound. 
Mike

bsongis, 2015-07-07:
At least not reproducible on simu

bsongis, 2015-07-07:
Which is your custom function line?

MIKE-DAILEY, 2015-07-07:
The switch is used for triple rates and was working fine on 2.0.17 unless I messed something up on the upgrade to 2.1.0. 
[image]
[image]
[image]

bsongis, 2015-07-07:
Do we agree it is SF5 and SF7 which should have the delay?

bsongis, 2015-07-07:
Oh yes you are right, the delay works with SA- but doesn't with SC-

MIKE-DAILEY, 2015-07-07:
SF5 is the mid position and should be impacted by the "Play Delay" SF7 is another switch used for playing the RSSI value when I want to during flight.

MIKE-DAILEY, 2015-07-07:
Correct, delay only on SA-.  Could this be related to my conversion problem?

bsongis, 2015-07-07:
No I have the same problem here with a blank model, and on the simulator

MIKE-DAILEY, 2015-07-07:
Great Bertrand, my testing effort helped!

bsongis, 2015-07-07:
Indeed, we need you in the beta user list

commit 9b7e913f (2015-07-07): Fixes #2434
closed 2015-07-07 by bsongis