opentx/opentx #2435: Possible mixer bug involving function x<0 and delays

th9xer, 2015-07-07:
Setup a mixer as per attached picture. With the mixer switched on when you rotate RS one way it has an effect on CH2 the other way it has no effect. This is expected. 

Now switch the mixer on, move RS to the side that has no effect, and switch the mixer off, you will see CH2 glitch as if RS was having an effect then decay to 0 (centre). No glitch occurs when switching the mixer on with RS in the same position.

The only factors that seem to matter is the mixer is switched, there is a delay, and x<0 or x>0 is used.

Tested using 2.0.17 on a Taranis A. Confirmed in both simulator and on actual set, (although only via the servo monitor screen). Checked for a duplicate report but couldn't find one. 

Also tested on 2015.07.07 nightly build of Companion 2.1, still occurs
[image]

th9xer, 2015-07-09:
Worth note.... I tried to circumvent the problem using a curve to represent the X>0 or X<0 function and the problem still exists.

projectkk2glider, 2015-07-10:
Reproduced, I will have a look...

projectkk2glider, 2015-07-10:
The problem is worse than I first thought. There is a general issue with the switched mixer lines that use slow up and down and at the same time use **offset or curve**.

Another example:
[image]

CH1 is a reproduction of the original issue. CH3 is my find, try to switch SA and observe glitches. Same issue is causing problems on both channels.

The culprit is this bit of code in `evalFlightModeMixes()`:
```c
      else {
        if (mode==e_perout_mode_normal) {
          swOn[i].now = swOn[i].prev = mixEnabled;
        }
        if (!mixEnabled) {
          if ((md->speedDown || md->speedUp) && md->mltpx!=MLTPX_REP) {
            if (mixCondition) {
              v = (md->mltpx == MLTPX_ADD ? 0 : RESX);
              apply_offset_and_curve = false;
            }
          }
          else if (mixCondition) {
            continue;
          }
        }
      }

```

The issue happens when `apply_offset_and_curve = false;` is executed. After that the mixer offset and curve are ignored. This causes glitches seen here.

Now the big question is, why was this code written as it is. I haven't found a good reason, but this bit of code was first introduced in https://github.com/opentx/opentx/commit/ce7adabfbeda0fe224a749572037a8210959793a#diff-6b6dcb460aa2b171dd65a02d813426eaR2441 and then expanded in 992e0f7ea475eed2315f01288abd4bee2879c37c and I am sure there was a reason for it. This part of the mixer code was then rewritten a couple of times but the basic disablement of curve and offset calculation still exists. 

@bsongis and @kilrah you were doing these changes, do you remember why?

projectkk2glider, 2015-07-10:
If I comment out the line 
```c
              apply_offset_and_curve = false;
```
Then both issues disappear. But I am sure this then breaks something else. I am afraid to touch mixing code, on account of it's complexity. We must in future take time and rewrite the mixer code in a more readable and more organized fashion.

bsongis, 2015-07-10:
We all agree on the necessity to rewrite this part of code :)
Did you execute google tests after removing this line
I wrote it, but I can't remember why ... I hope there is a non regression test ...

projectkk2glider, 2015-07-10:
Gtests pass. I can't decide if that is good or bad :worried:

bsongis, 2015-07-10:
No one test fails on 9x board!

projectkk2glider, 2015-07-10:
My current thinking is to leave this one alone until we decide to do a complete rewrite of the mixer code. Too many possibilities that we break something else while fixing this.

bsongis, 2015-07-10:
Right. Before fixing anything we will have to answer a simple question: is slow applied to the input of the mix, or the whole mix. I would be in favor of the whole mix. So the slow code should be moved to the very end of the block, it will simplify things ...

bsongis, 2015-07-10:
Today slow applies to the input, but when completely disabled, the offset / trim / curve is not applied (while it's applied during the whole slow transition), this is not good I think.

bsongis, 2015-07-10:
@JivaroFAD This issue will have to be taken into account in our refactoring!

projectkk2glider, 2015-07-10:
I also think that slow should be at the very end of mixer.

kilrah, 2015-07-10:
> is slow applied to the input of the mix, or the whole mix. I would be in favor of the whole mix. 

Won't we have a problem if the same input is used for multiple mixes that have different weights? The slow would have to take the output into account if we want the "ends" to match, and that isn't really predictable unless we do recursive calculations we wanted to avoid...

JivaroFAD, 2015-07-10:
+1 for applying slow and delay on output.

I found the same issue while working on differential, but I though in the moment that I had broken something and later I forgot about it :(

> @JivaroFAD This issue will have to be taken into account in our refactoring!
I agree. So, in the new mixer loop, we will have something like this (from raw to ouput) :
- offset if "OffsetOnInput" = true
- curve (raw + trim) or diff(raw) + diff(trim)
- weight
- offset if "OffsetOnInput" = false
- delay
- slow

pindeslandes, 2015-08-07:
I don’t know if this is related but there is that funny issue about slow and delay:

[image]

Having set the four mixers above:
1-When SAv and moving SF:
Only slow works on channels 1 and 2 while slow and delay work as expected on channels 3 and 4. Logically, channel 1 and 3 (resp. 2 and 4) should output the same.
2-When moving SAv on => off (no matter the position of SF), only slow works on channel 1 and only delay works on channel 2.
3-When moving SAv off => on (no matter the position of SF), only delay works on channel 1 and only slow works on channel 2.

kilrah, 2015-08-07:
Currently slow and delay are not properly supported when the line may be disabled by a switch, been like that for ages.

pindeslandes, 2015-08-07:
For ages, sure, as it was already there in V1. But as the present issue is only one month old, and as it involves switches, I felt like I could allow myself to add my ten little pence.

bsongis, 2017-02-15:
Will be part of the mixer rework done by @JivaroFAD

closed 2017-02-15 by bsongis