opentx/opentx #2437: Add ST7920 LCD support for M128 / 9X Board

exwhyzed01, 2015-07-08:
Hi,

I'm making a homemade RC transmitter, and am trying to implement an ST7920 board.  I see that its been included, in Companion 2.1 for the 2560 Board.  Would it be possible to add this functionality for Atmega128 boards as well?  Thanks for looking into it!

bsongis, 2015-07-08:
No problem

exwhyzed01, 2015-07-11:
Thanks, I appreciate it!  opentx makes rc hobbying complete by giving people like me the means to not only create RC vehicles, but also the means to control them.  yaay!

exwhyzed01, 2015-07-14:
On a whim, I decided to copy and paste a line from the 'MEGA2560 board' section, line 1284, of file
companion/src/firmwares/opentx/opentxinterface.cpp

into the '9x board with M128 chip' section, such that it is now line 1168

the line reads:
addOpenTxLcdOptions(firmware);

I followed the instructions on building a companion in Linux but i'm currently snagged at Qt 4.8.4 which doesn't want to install on my machine for some reason.

I'll keep trying things out, but if someone gets to it first, I won't mind :)

exwhyzed01, 2015-07-24:
After much painstaking attempts, i can not get past Windows and Linux cmake compiler errors, argh.  I give up, and will wait for the OpenTX solution!

bsongis, 2015-07-24:
Were is your fork so that I can have a look?

exwhyzed01, 2015-07-24:
Hi bsongis,

My github fork is [here](https://github.com/exwhyzed01/opentx)

I looked at the pin assignments in [board_stock.h](https://github.com/opentx/opentx/blob/master/radio/src/targets/stock/board_stock.h) and everything seems to fit onto an Atmega128 development board using the same pins, with some pins to spare, so i didn't make any changes here, or anywhere else.  Thanks in advance for taking a look!

closed 2019-08-19 by kilrah