opentx/opentx #2440: A way to select 2 soundpacks within 1 language (male/female?)

fraternl, 2015-07-08:
I am considering to make a soundpack with a (real) female voice in my own language. I just now became aware that someone already made a soundpack in my language.

This soundpack is however with a male voice.

It would be nice if an alternate voice-pack could be used within the language...

kilrah, 2015-07-08:
Just rename the folders?

fraternl, 2015-07-08:
I have my radio not here...
The language is not hardlinked to the 2 character language code?

Then it's not necessary...

kilrah, 2015-07-08:
It is hard-coded, yes.

fraternl, 2015-07-08:
But when I select "Nederlands". it will play the sounds of the /SOUNDS/nl/ folder.
Maybe only a checkbox for "alternate sounds" and then it will play the sounds out of the /SOUNDS/nla/ folder???
The same for the other languages
This way it would also be able to switch between Amber and Standard.

What I'm proposing has a very minor impact on the code.

(I know there's no "Nederlands"... I'm working on it)

kilrah, 2015-07-09:
If I want to change sounds I just have an "en" folder and say an "en_male" one... If I want to switch I'll rename "en" to "en_female" and "en_male" to "en". 

But then again I don't see why I'd change packs all the time, I try a few, choose one I like and once I'm done testing it stays there.

fraternl, 2015-07-09:
true true true....
But like I said.... It could be made with a minimal impact on the code...

Maybe a late milestone or close it....??
I still need to make that good female voicepack.....

kilrah, 2015-07-09:
> It could be made with a minimal impact on the code...

Go ahead then ;)
But then why only 2, because some people will want to swap between 3 or 4... ;)

fraternl, 2015-07-11:
That could be easily accomplished by selecting a suffix..
a to a
Default, there's no suffix.

closed 2016-11-27 by fraternl

bsongis, 2016-11-28:
Why do you close this one, I do like the idea, just needs some time to come
with a nice solution...