opentx/opentx #2441: Telemetry - setting units degree F removes prec field in companion & simulator

Mow06, 2015-07-08:
[image]
[image]

kilrah, 2015-07-08:
normal, both are incompatible.

closed 2015-07-08 by kilrah

Mow06, 2015-07-08:
Don't understand. Why is degree F the only one without a prec field?

Mow06, 2015-07-08:
Don't understand.

What "both" are incompatable.

You create only one sensor with degreeF, prec field disappears. What is that not compatable with

  ----- Original Message ----- 
  From: Andre Bernet 
  To: opentx/opentx 
  Cc: Mow06 
  Sent: Wednesday, July 08, 2015 10:16 PM
  Subject: Re: [opentx] Telemetry - setting units degree F removes prec field in companion & simulator (#2441)

  normal, both are incompatible.

kilrah, 2015-07-08:
With the the way precision is calculated changing it would break the °C to °F conversion.  That's why it's hidden in this case.

Mow06, 2015-07-08:
OK. Thanks for the explanation.