opentx/opentx #2443: Adding Nederlands (Dutch) as a selectable language

fraternl, 2015-07-09:
Adding Nederlands (Dutch) as a selectable language

I've been discussing this with Bernard in a closed ticket.
It's more appropriate to have its own ticket

Dutch soundpack
    http://hmvc.eu/SOUNDS_nl.zip

A patched tts.en.cpp to be used for the nl-language
   http://hmvc.eu/tts.nl.cpp

I'm still working on the nl.h.txt file  (it's time consuming)

bsongis, 2015-07-09:
https://github.com/opentx/opentx/commit/d0bee7d66679b28f4e86ff623b64b751abb75fea

closed 2015-07-09 by bsongis

fraternl, 2015-07-13:
Could you add this for the Dutch translation?
This is probably not the final one.
Is this a problem???

[image]