opentx/opentx #2446: [Bug][OpenTX 2.0.17][Companion] No RSSI alerts

LapinFou, 2015-07-09:
Hi folks,

When I'm using the F4 button under the simulator (telemetry test), if I'm changing the RSSI nothing is happening. No beep, no alarm, no sound announcement.
I don't know if it is only a simulator bug or if this can also somehow impact the OpenTX firmware itself.

projectkk2glider, 2015-07-09:
Which version?

LapinFou, 2015-07-09:
2.0.17
I forgot to mention a very important point in order to reproduce the issue: you must launch the simulator from the model edition window.
If you launch it from the main window, the problem doesn't appear.

projectkk2glider, 2015-07-09:
Confirmed (even in next), the RSSI value changes, but the warning does not sound.

projectkk2glider, 2015-07-09:
I've found the problem. We use static variables that work first time the simulator is started, but on consecutive starts the initial value is not set again. The culprit here is `alarmsCheckTime` in `telemetryWakeup()`.

This issue is not affected by the way the simulator is started. It works the first time and fails on the second time.

closed 2015-07-10 by projectkk2glider

LapinFou, 2015-07-11:
That was a fast fix!!!
Thanks.
:)

commit 19567b09 (2015-07-10): Fixes #2446: some function static variables were dependent on continuity of g_tmr10ms value between simulator sessions.
commit 87d90e39 (2015-07-10): Fixes #2446: some function static variables were dependent on continuity of g_tmr10ms value between simulator sessions.