opentx/opentx #2447: [Enhancement][Companion 2.0.17] More telemetry parameters under the simulator

LapinFou, 2015-07-09:
Hi folks,

It would be very nice to have more parameters available when hitting the F4 button under the simulator.
Telemetry params are missing:
- batt:Radio battery voltage (this one is really missing)
- fuel
- cell 1, 2, 3.. 6

projectkk2glider, 2015-09-22:
From #2915:
>Why we don't have the simple possibility to enter until 6 cells values separated by a semicolumn in CELLS and a converter will crunch this string to create the right s.port format internally. And at the other side, getValue able to have a second optionnal parameter, an indice to get the simple value : eg getValue("Cels",4) to get the value of Cels number 4.
GPS-LL, 2 values, "longitude;latitude", GPS-T, 6 values, H;M;S;Y;M;D. Something like that..

fred99, 2015-09-22:
A better last version of my request:
Why we don't have the simple possibility to enter a string: until 6 cells values separated by a semicolumn (a value can be in PREC2) in CELLS field and a converter will crunch this string to create the right s.port format internally. And at the other side, getValue able to have a second optionnal parameter, an indice to get the simple value : eg getValue("Cels",4) to get the value of Cels number 4.
GPS-LL, 2 values: longitude;latitude, GPS-T, 6 values: Y;M;D;H;M;S. Something like that.

projectkk2glider, 2016-01-04:
Many improvements done in #3124, we (@dsbeach :wink: )  must check if anything is still missing.

fred99, 2016-01-05:
Very nice job, the telemetry simulator is becoming friendly and easy to use.
Thanks a lot for this big improvement.

LapinFou, 2016-01-26:
With the latest version of Companion 2.1.x, all my request have been implemented in #3124.
Guys, you are the best!! :smile:

closed 2016-01-26 by LapinFou