opentx/opentx #2448: Companion:  Check firmware compatibility gives me 2 error messages

fraternl, 2015-07-10:
An observation, maybe not a bug.

Last weekend I flew with 2.1.
Today, the first thing I did was update the firmware to today's daily (was very pleased to see  the language Nederlands added).

I then loaded the models from my radio into companion.
I noticed that the receivers on all models were turned off in companion (maybe this is normal?).

I saved my settings on the radio to be able to revert to this situation and I wrote all the settings to the radio to see if my radio would have the receiver set to off.

Companion came with this error:
[image]
[image]

I checked the radio afterwards and the receiver settings were still in the radio.

Then I repeated the procedure. (the radio has the settings that were loaded with today's companion).
The same thing happened....

I don't get these error messages if I turn off "Check Firmware Compatibity"
Can you tell me if it's normal that radio settings don't appear in Companion 2.1?

[image]

EEPROM
[image]

fraternl, 2015-07-10:
The settings in my radio did get compromised after this procedure. The "Check FW compatibity" prevented the writing to the radio, I now realize.
I can now continue to turn the radio on for all models and repeat the procedure.....

Maybe all this is normal because there was an eeprom change.

Should I check more than just radio settings?

kilrah, 2015-07-10:
> noticed that the receivers on all models were turned off in companion (maybe this is normal?)

No it's issue #2425.

fraternl, 2015-07-11:
I went back to 2.0.17 loaded my last 2.0.17 and went through the complete upgrade procedure.
That's issue is solved.

It did give me a chance to do the whole procedure again.
This time I filled all telemetry parameters using Companion....

In the process I discovered a little bug for which I will now create a ticket  :-)

kilrah, 2015-07-11:
So is there still a problem with the check compatibility feature?

fraternl, 2015-07-11:
I have definitely upgraded to today's ROM
Companion 2.1 still give me a warning if I write the settings to the radio with the option "Check FW Compatibilty"

closed 2016-11-27 by fraternl