opentx/opentx #2453: Sounds are broken again

fraternl, 2015-07-11:
The units metres and volts are not being pronounced anymore
They were working fine last Sunday.

To make sure I just downloaded the soundpack of 2015-07-11 and renamed the /SOUNDS/en to /SOUNDS/ena
Then I extracted the ZIP directly to the SD-Card....

Maybe more units are off....

bsongis, 2015-07-11:
Which pack did you download? Which language?

bsongis, 2015-07-11:
> Already too many hours testing while it's beatiful weather (not that common where I live)

Only hours for you ;)

MIKE-DAILEY, 2015-07-11:
I'm on build #33 now with the US English voices and I'm getting the correct voltage voice for A1, but it worked fine for me on build #31 too.

rdeanchurch, 2015-07-11:
There are differences in 2.0 and 2.1 for sound packs.  Have you downloaded a 2.1 sound pack?
Kilrah made one and there is one on RcGroups, a modified Amber pack.

bsongis, 2015-07-11:
May I have the modified Amber pack link? I will add it to the server ...

bsongis, 2015-07-11:
Please reopen if needed, but I don't think there is an issue with the latest sound packs!

closed 2015-07-11 by bsongis

rdeanchurch, 2015-07-11:
I didn't do it frater did
 http://www.rcgroups.com/forums/showthread.php?t=2451551&pp=100&highlight=Amber#post32078692
post #52.  It took care of the few system sounds I was using that were wrong.

bsongis, 2015-07-11:
Thanks, I copied it to the server with others

fraternl, 2015-07-11:
I took the American-English pack.
I assume it's the one I should take as reference

If I do a play value of the altitude I will only get "zero", not "zero metres"
The same for my Dutch pack, which was working last Sunday.

I am still looking for some user error of mine, but I can't find it.

bsongis, 2015-07-11:
Prompt 132 "meters" should be played here.

fraternl, 2015-07-11:
0131.wav says "metre" and 0132.wav says metres.

I checked it using SD-card manager, do it can't be an invalid sound file.

Are you saying it's pronouncing metres on your system?

bsongis, 2015-07-11:
Yes it does, I just checked with the american pack.

bsongis, 2015-07-11:
Amber pack also good, all units tested again.

fraternl, 2015-07-12:
It now turns out I've been checking a model where the variometer wasn't setup properly.
Everything's there though, buy something must have gone wrong there.

I just checked another model and there it worked.
The model I was testing all this time is still at the club. I need to give the sensors a proper testing.

I also tested this working model when it's turned off, but then it still says metres....

fraternl, 2015-07-12:
I checked the model settings of that model that isn't there.
The special function which had the "Play Value     Alt"   was still at "Play Value    ----"

Because it did speak the value "zero", I never thought of checking the special function....

I leave it up to you to decide if "Play Value  ---" should say nothing or "zero".
I would prefer if it said nothing.

It's "garbage in => garbage out" of course.

commit 00c62b13 (2015-07-12): Fixes #2453