opentx/opentx #2454: Trims not correct,  showing +/-125%  instead +/-25% at one screen

hrenz, 2015-07-11:
openTX V2.11 latest next
Trims value at channels monitor is ok +/-25%
but
at show trims aktiv, no extendet trimm aktiv  
shows +125% / -125%
[image]
[image]
[image]

hrenz, 2015-07-11:
sorry thing has cleared, this mean:
At channel monitor is at % or at PPMus
At the other screen is +/-500 steps   with extrafine=1 fine=2  mid=4 steps  and so on.

But:
If a trim is  at 125  (= 25%) and I push the same trim button again
it runs over 125 to max 500
(no extendet trim is aktiv!) can that be ok?

kilrah, 2015-07-11:
Again please explain PRECISELY what you're doing. I just tested trims and they only go to 500 if extended trims are on.

hrenz, 2015-07-11:
Hy kilrah,
no extendet trim aktiv!
push a trim button, hold it, runs to 125,  ok its 25%
then push same trim button again, hold it, and its runs from 125 to 500  (=100%)

kilrah, 2015-07-11:
And I can't reproduce that either on the radio or on the simu. 
So please say exactly what version you're using, whether it's the radio or the simu, and what you're doing coming from a new model to reproduce the problem.

hrenz, 2015-07-11:
test with openTx V2.10 latest from today 
at Simu with Modell1 at start up
-----------------------------------------------
now I have generated a new Modell3
there I can't reproduce the "bug"
there it works well as it should 
----------------------------------------------

closed 2015-07-11 by bsongis