opentx/opentx #2456: GPS Lat/Lon logging with new 2.1 telemetry

Bamfax, 2015-07-11:
It seems the GPS latitude and longitute are not logged correctly to the SD logfile. When logging the GPS sensor, it is logged as one field with an odd 5- oder 6-digit integer value, not separate lat/lon fields.

I write some more details here:
http://openrcforums.com/forum/viewtopic.php?f=45&t=6887&p=99028#p99028

I hope I have not missed something obious.

Thank you for 2.1 and for fixing this.

Cheers,
Olli

bsongis, 2015-07-12:
Would you please test the other non standard fields? I am afraid of Cells, Date and perhaps others. Thanks

closed 2015-07-12 by bsongis

Bamfax, 2015-07-17:
With OpenTX 2.1.0 it's now logged fine. But it caused a subissue: The now new Latitude field does not log its field name at the beginning of the logfile. So the log header shifts every following field name by one and the the last field name is missing.

bsongis, 2015-07-18:
Yes please a new issue, thanks!

commit b4364e13 (2015-07-12): Fixes #2456