opentx/opentx #2457: Companion 2.1: In telemetry the tick box for precision is not making the digits it should

fraternl, 2015-07-12:
From what the telemtry system autoprovisioned from my setup it seems the field in companion should be 0, 0.0 , 0.00, 0,000

You can also put 2 or 3 in there. I don't know if it's supposed to be possible, but I would think not.
The up/down box besides it will change the value of the field like any other value.

bsongis, 2015-07-12:
Sorry, but I don't understand what you mean.

kilrah, 2015-07-13:
Precision = how many of the value's digits are after the decimal point. Can be 0, 1 or 2 and that's what companion displays and accepts. Just tried to misuse it and it seems to work as intended here.

closed 2015-07-13 by kilrah

fraternl, 2015-07-13:
O.k.
But to be clear... It accepts both?
When I import the settings in companion it says 0 0.0 or 0.00

kilrah, 2015-07-13:
There's only one way to set it. But when you change the resolution of the value, the resolution of the offset follows.