opentx/opentx #2460: Companion 2.1 Telemetry sensor range incorrect

pinkywafer, 2015-07-12:
Running 2.1 on Taranis (not plus).
I set up a quad using NAZE32 and D4r-II. Telemetry sensors were detected on the tx, and all seems fine. I have adjusted a couple of the "range" settings for the sensors (on the TX) : VFAS range 25.5.  RPM range 1279.0.  These give the correct expected readings on the TX.  When imported into companion, while VFAS range shows 25.5, RPM range shows 24.6

bsongis, 2015-07-12:
Which range do you mean?

pinkywafer, 2015-07-12:
on the sensors page. Its labelled range in companion, and ratio on the tx. 
[image]

[image]

commit 385536ab (2015-07-13): Fixes #2460
closed 2015-07-13 by bsongis