opentx/opentx #2461: Firmware Simulator 2.1 Pots

pinkywafer, 2015-07-12:
If I load firmware simulator 2.1, there are no controls for pots
If I hit simulate from within companion, the controls for pots appear as normal at the top by the switches.

projectkk2glider, 2015-07-12:
I am not entirely sure I understand you. Do you mean that pots are missing from Standalone Simulator?

pinkywafer, 2015-07-12:
yes.  See attached
[image]

MIKE-DAILEY, 2015-07-12:
Looks like this for me on build #33.
[image]

bsongis, 2015-07-12:
Did you configure your pots in General Settings? If you set type None for all of them you will get no pot in the simulator.

pinkywafer, 2015-07-12:
Yes, pots are configured as pot with detent. They are there in the simulator within companion, but not on the standalone simulator

bsongis, 2015-07-12:
Ok understood.

commit 5c3cfd9c (2015-07-12): Fixes #2461
closed 2015-07-12 by bsongis