opentx/opentx #2462: Timer3 Missing

Juemo, 2015-07-12:
I running 2.1.0 from 2015/07/12 on Taranis (B01) but Timer3 is missing.

hrenz, 2015-07-12:
Look at makefile:
At the makefile, line 273 you can select 3 Timers,
but at line 892 there  is fix Timers = 2 
so you got only 2 Timers  
Change at line 892 Timers = 3 and it will work.
its a bug  since some days.

bsongis, 2015-07-12:
No. It's only a missing option on the compilation server, I will add it right now.

closed 2015-07-12 by bsongis

kilrah, 2015-07-12:
Why make an option out of this? I'd have included for all ARM boards in the makefile.

bsongis, 2015-07-12:
Because most of people don't need 3 timers, including me, I prefer not having it in the menus. The same than heli / gvars etc.

fred99, 2015-07-20:
After selecting the option timer 3, reflashing the radio and activating the timer 3, it's not in the main page of the Taranis. However, we can get it in telemetry screen (page long) if we add it in page 13/13.
Is it an implementation choice because of the limited room in the main screen ?

kilrah, 2015-07-20:
Yes, people who wanted a 3rd timer wanted to use it mainly to count total model flight time over long periods like the whole season, i.e. they don't need to see it all the time.

fred99, 2015-07-20:
Ok, I share this information on http://frskytaranis.forumactif.org.
Thanks for your quick answer.

commit 0a3bcd27 (2015-07-12): Fixes #2462