opentx/opentx #2463: Firmware 2.1 LUA telemetry screen issue

pinkywafer, 2015-07-12:
I'm not certain if this is an issue, or intentional:
I've edited my telemetry script, and it will run fine on 2.1 when selected as a custom telemetry screen, but when switching to the telemetry screen, it is impossible to move off it.  ie. before (or currently when displaying a nums/bars screen), pressing exit returns to normal screen and pressing page switches to the next telemetry screen.  Neither of these work. It is as if I had always called killEvents(event).

Is this intentional?

bsongis, 2015-07-12:
Would you provide a link to your screen and eeprom?

pinkywafer, 2015-07-12:
eeprom and lua are at:
https://www.dropbox.com/sh/hl8omk9mvee4iqy/AACyXQQXZQI9RcB8L86HVw_Ia?dl=0
using model: fpv250

bsongis, 2015-07-13:
Would you provide the eeprom after conversion? I see no sensor there.

pinkywafer, 2015-07-13:
sorry wrong file uploaded!.  I have deleted this and uploaded my 2.1 version eepe at the same location

pinkywafer, 2015-07-13:
I have also tried a very simple telem script:

    local function run(event)
	   lcd.clear()
	   lcd.drawText(7,2,event,0)
    end

    return { run=run }

this works - refreshes every cycle and displays just the numeric value of `event` but it has the same fault - once started, the screen can't be closed - the number will briefly (until the next cycle) display the event numbers of keypresses, but exit does not exit, and page does not move to the next page, even though I have not used `killEvents(event)`

hope that clarifies it a bil

bsongis, 2015-07-13:
Reproduced. This is not normal

bsongis, 2015-07-13:
I uploaded the modified script there:
http://downloads-21.open-tx.org/taranis-sd/SCRIPTS/TELEMETRY/

bsongis, 2015-07-13:
I start a new compilation right now, I would really need it's tested before X9E goes to production, thanks!

closed 2015-07-13 by bsongis

pinkywafer, 2015-07-13:
Where will I find the new binary? Then I will test shortly

bsongis, 2015-07-13:
You download the latest Companion:
http://downloads-21.open-tx.org/companion/
Then you download again your firmware and flash it!

pinkywafer, 2015-07-13:
Thanks - Looking good now, and thanks for the catch clause, I can see it's now needed as the field would not be present if the sensors weren't set up whereas before the fields were always present.

I had to check the sdcard.h code also, as I was getting 'No Scripts on SC Card'. I see that the path has now changed from SCRIPTS/TELEM to SCRIPTS/TELEMETRY

Seems to be working fine now.  Thanks

commit 5fb12915 (2015-07-13): Fixes #2463 - Code review needed, thanks!