opentx/opentx #2464: New translation "Nederlands"

fraternl, 2015-07-13:
I created http://hmvc.eu/nl.h.txt from en.h.txt to be used for the translation to Dutch

bsongis, 2015-07-13:
Please check string lengths, you cannot write this for example:
#define LEN_VBEEPMODE          "\005"
#define TR_VBEEPMODE           "Stil""Alarm""NoKey""Alles\0"

closed 2015-07-13 by bsongis

fraternl, 2015-07-13:
OK, I will

But how should I see this in the original en.h.txt????

#define LEN_VRENAVIG           "\003"
#define TR_VRENAVIG            "No REaREb"

mhotar, 2015-07-13:
"No REaREb" is the same like "No ""REa""REb", this is something that needs to be changed for better readability.

fraternl, 2015-07-13:
I have updated the file now.....
http://hmvc.eu/nl.h.txt

kilrah, 2015-07-13:
Some lengths are still off... the LEN is in octal (base 8), not decimal. So for example TR_VPERSISTENT either has 2 chars too many on each item, or its LEN should be 2 more.

kilrah, 2015-07-13:
Also you should ideally have a build environment set up so you can build the simulator. With the thing above fixed as well as 3 missing " some strings seem to be too long, some only on small screen platforms and some also on Taranis. You can use the TR() macro to define separate strings for both, but it's hard to do without being able to test in "real life".

[image]
 
[image]

[image]

[image]

I've committed the last file and the couple of fixes that result in the screenshots above.

fraternl, 2015-07-13:
I didn't know it was octal (I should have). Did know that \0 is 1 char.
I have a corrected file for download, but I think it's almost impossible to get it right...
English words are generally much shorter than Dutch words

kilrah, 2015-07-13:
Time to get creative with abbreviations :)

fraternl, 2015-07-14:
I downloaded the new firmware to my radio and realized I didn't understand how it's implemented now.
It's not in the radio firmware, but only in the simulator....

Could you give me some global directions for making a development environment?
Is it in Windows or Linux?
I assume in Windows as the target is in Windows.
In Linux I don't use the desktop versions
Never developed in Windows. Currently only in Linux and a long time ago for MS-DOS.

If mismatches in the "Dutch part" have no influence on the behaviour of the rest of the system you could just "make it live" for the radio itself. I have the radio now with me and can check everything the next morning.
I am aware that OpenTX is not "Taranis only"
I will also examine and compare the German file.
German has the same "problem" as Dutch using much longer words for the same thing.

Is that OK by you?

kilrah, 2015-07-14:
The radio firmware will have it too, but something was missing that I fixed in my last commit so for now it wasn't "enabled" neither in the simulator nor on the radio. 
Next 2.1 beta release will have NL present both in the radio and simu, but as these are only made once every few days feedback is kind of slow.

Build instructions are in the project's github wiki.

fraternl, 2015-07-14:
If it is build could you fetch a fresh one from  http://hmvc.eu/nl.h.txt
I think it should be usable now

kilrah, 2015-07-14:
It also reverts at least 2 of the fixes we did... Would really be good if you could use the git tools ;)

fraternl, 2015-07-16:
OpenTX 2.1.0 has come out now and none of the translations are implemented. Is that correct?

kilrah, 2015-07-16:
I just downloaded an NL firmware from companion and flashed it, and it is in NL.