opentx/opentx #2465: Feature request: Companion 2.1 Telemetry Script

pinkywafer, 2015-07-13:
On the companion, Telemetry tab, for each screen, if Script is selected, is it possible to have a dropdown to select the script to use (assuming SD card is synced)?

AndreasAusD, 2015-07-17:
on the simulator it is already working right, only in companion there is no way of selecting a script.
also this information is lost whe there is a script assigned on the radio and the models are downloaded from the radio to companion and uploaded back to the radio again

bsongis, 2015-07-18:
Would someone test this one before the release, I did a small refactoring, I would like to be sure I didn't break anything ...

projectkk2glider, 2015-07-18:
One bug found, maybe not even related. When I defined bars in Companion like this, I get empty bars screen in simulator (both ranges are 0-0). 
[image]

bsongis, 2015-07-18:
You should use another issue, it's not at all related!

projectkk2glider, 2015-07-18:
Done, issue #2491

closed 2015-07-18 by projectkk2glider

AndreasAusD, 2015-07-18:
will this go to the nightly builds for testing ? and when ?

bsongis, 2015-07-18:
Yes, nightly. I will wait befinning of next week for 2.1.1

projectkk2glider, 2015-07-18:
The new nightly will be build in about 10 hours from now.

commit 90ebaf2d (2015-07-18): Fixes #2465 - A little bit refactoring done while fixing this one
commit a12f0fdf (2015-07-18): Fixes #2465 - A little bit refactoring done while fixing this one